verilator/verilator #2908: New Warning Message for

qrqiuren, 2021-04-28:
> Thanks for taking the time to report this.

> What would you like added/supported?

```systemverilog
// File: dly_warning.sv

// verilator lint_off ASSIGNDLY

module dly_warning (
    input logic a_in,
    input logic [2:0] delaycw,

    output logic a_out
);

timeunit 1ns;
timeprecision 1ns;

time dly;

assign dly = 5 * delaycw;
assign #dly a_out = a_in;     // Warning ASSIGNDLY in this line has turned off

endmodule

```
In the above code, when I use `verilator --lint-only -Wall` to it, it shows the following message.
```
➜  sv verilator --lint-only -Wall dly_warning.sv
%Warning-UNUSED: dly_warning.sv:13:6: Signal is not used: 'dly'
                                    : ... In instance dly_warning
   13 | time dly;
      |      ^~~
                 ... For warning description see https://verilator.org/warn/UNUSED?v=4.202
                 ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
➜  sv 
```

However, `dly` is obviously used in the delayed assignment.

Although I know that Verilator is ignoring delays, I still want to use my files in other simulators. My suggestion is (1) to ignore UNUSED warning in this kind of situation or (2) make a new kind of warning message for this situation, so that people can turn off directly.

> Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

Sorry, I don't have any other openly licensed linting software.

> May we assist you in trying to fix this yourself?

I hope so, but I don't have much time recently.

wsnyder, 2021-05-03:
I agree it should not warn in this case.  Currently the delay's AstVar doesn't get out of the parser, continuous_assign in verilog.y just leaks it (oops). instead it should propagate the delay down, and the linter can then see it.

Good small project if someone would like to look at it.  Doing this also helps getting closer to delayed statements working.

adhason, 2022-03-21:
I would like to try to look at it if that's okay.

wsnyder, 2022-05-15:
Note #3363 when merged will fix this.

closed 2022-08-30 by wsnyder