verilator/verilator #2911: Commit c1e8337: causing explosion in V3Width Runtime

ngraybeal, 2021-04-29:
I'm in the process of upgrading from version 4.028 to 4.200 and noticed an explosion in time performing V3Width. The elapsed time went from roughly 3 seconds to over 2100 seconds. Using callgrind, I was able to trace the cause back to commit c1e8337fc1b36652328eb00a5dd256d5c9a33691: "Support pattern assignment to dynamic arrays". When i reverted this commit, I did indeed see the runtime fall back to the ~3 seconds and seemingly had no effect on my build/compile/run as all were successful. 

**Was this performance hit expected from the new feature?** 

Deeper debug details:
Note: "-Wno-WIDTH" is currently used. 
I did try running against latest version 4.202 and saw similar high runtimes. 

WidthVisitor::visit(AstConcat*) executes a large number of instructions with WidthVisitor::userIterateAndNext and WidthVisitor::iterateCheckSizedSelf being the main factors (roughly 25% and 75% respectively). The addition of userIterateAndNext(vdtypep, WidthVP(SELF, BOTH).p());   looks to create a subtree recursion that can lead to a bunch of executed instructions if the sub nodes are AstConcat types, which seems to be the case in my design. 

Sorry I don't have a test case, as I'm still getting familiar with Verilator constructs and pinpointing the source code that hits the scenario above.

wsnyder, 2021-04-29:
This certainly was unintended, if you can get it to a test case it would be helpful, otherwise I'll have to see if I can intuit what it's doing.

ngraybeal, 2021-04-30:
Thanks, I'll work on getting a testcase in parallel. Just wanted to start the conversation in case this might have been known or obvious.

wsnyder, 2021-06-04:
Sorry forgot to acknowledge earlier this does seem a bug.  If you'd like to debug it further it would be appreciated, otherwise will get to it when we can.

wsnyder, 2021-06-07:
I'm not sure but suspect the fix I just committed will solve this - can you please confirm?

ngraybeal, 2021-06-07:
I can confirm this has solved the issue I was seeing. Thanks for the support!

closed 2021-06-07 by wsnyder