verilator/verilator #2913: Using multiple threads results in dramatic slowdown

skeetor, 2021-05-01:
I built verilator from the latest master on running debian 10. I was using a simple module and tried to run it using multithreading, but the performance drops dramatically.

--no-threads : 1.89565s
--threads 1: 3.15899s
--threads 4 : 7.63842s

Now I wonder why, because I expected an increase of performance, but not such a dramatic slowdown. I was using other modules which are more complex and the result is pretty similar so I created this small module to see if there is some problem with my code but it seems it's not related to that.

I also used the numactl but there is no difference. So I wonder if I did something wrong, or if there is some proble with verilator.

Module:
`module our(
        input clock, 
        output reg [31:0] fn = 0);

   reg [31:0] fn1 = 1;

   always @(posedge clock) begin
      fn <= fn1;
      fn1 <= fn1 + fn;
   end

  initial begin
    $display("Fibonacci numbers:");
  end
endmodule `

And using this cpp code:
`#include <iostream>

#include "Vour.h"
#include "verilated.h"
#include <chrono>

struct profiler
{
    std::string name;
    std::chrono::high_resolution_clock::time_point p;
    profiler(std::string const &n) :
        name(n), p(std::chrono::high_resolution_clock::now()) { }
    ~profiler()
    {
        using dura = std::chrono::duration<double>;
        auto d = std::chrono::high_resolution_clock::now() - p;
        std::cout << name << ": "
            << std::chrono::duration_cast<dura>(d).count()
            << std::endl;
    }
};

#define PROFILE_BLOCK(pbn) profiler _pfinstance(pbn)

int main(int argc, char **argv, char **env)
{
	volatile uint64_t loop = 0;
	Verilated::commandArgs(argc, argv);
	Vour *top = new Vour;

	{
		PROFILE_BLOCK("Timer");

		while (!Verilated::gotFinish())
		{
			if (top->clock)
				loop++;

			top->clock ^= 1;
			top->eval();
			if(loop >= 100000000)
				break;
		}
	}

	std::cout << "Loops: " << loop << std::endl;
	delete top;
	return 0;
}
`

wsnyder, 2021-05-01:
Multithreading will only show speedups on much larger designs. In small designs the communication between cores will be much larger than leaving it on one core.

closed 2021-05-01 by wsnyder

gergoerdi, 2021-05-02:
@skeetor forgot to link to [my original Stack Overflow question](https://stackoverflow.com/q/67335512/477476), where I stumbled upon this problem. That one is with a "real" (albeit simple) design that generates a VGA signal showing a bouncing ball. However, originally I came upon this behaviour (that multithreading slows down simulation tremendously, instead of speeding it up) in the context of a much larger design that implements a full Space Invaders arcade machine (with 8080 core and all); full (Clash-generated) Verilog of 20468 lines available on request (but I guess its specifics won't make much of a difference).

So in light of all this, @wsnyder can you characterize the circuits that *should* be able to benefit from multithreaded simulation?

wsnyder, 2021-05-02:
That sounds large enough to get some benefit but mileage varies. Can you make a Gantt report with 2 threads (see docs) and post the output please?