verilator/verilator #2914: 2-dimension associative array

myftptoyman, 2021-05-01:
Verilation version:  4.201 devel rev v4.200-105-gf4462952

The two dimension associative array compile get error for _ctor_var_reset 

Example:

```
module top;
  logic [31:0] acc_arry [logic [31:0]][logic [63:0]];
 initial begin
    acc_arry[5][9] = 9;
    $display("%d", acc_arry[5][9]);
  end
endmodule
```

It get compiler error:

Vtop_top__Slow.cpp:79:53: error: no match for 'operator=' (operand types are 'VlAssocArray<long unsigned int, unsigned int>' and 'IData' {aka 'unsigned int'})
   79 |     __PVT__acc_arry.atDefault() = VL_RAND_RESET_I(32);
      |                                                                                                 ^

It can fixed by ***__PVT__acc_arry.atDefault().atDefault()***

[emitVarReset] (https://github.com/verilator/verilator/blob/422c076fec475a0cdfee613c6d664e97fe565cd5/src/V3EmitC.cpp#L1741) 

maybe have get wrong data type in recursive search

wsnyder, 2021-05-01:
It looks like you're on the right trail, perhaps you could propose a pull request for the test case and patch?

wsnyder, 2021-05-03:
Looked at this and it was straightforward, fixed towards 4.204.

closed 2021-05-03 by wsnyder