verilator/verilator #2920: Verilator 4.202 uses Makefile syntax which doesn't work on RHEL/CentOS 7

imphil, 2021-05-04:
https://github.com/verilator/verilator/commit/6a189a1f8174bfa7c6829f1c0611092c5dbf2bc6, a fix for https://github.com/verilator/verilator/issues/2834, released in Verilator 4.202, introduced Makefile syntax which is incompatible with GNU Make < 4 shipped with CentOS/RHEL 7, causing `make test` and other builds to fail with an unhelpful error message.

Exemplary error message:

```
make test
# ... snip ...
ar: @Vt_a1_first_cc__ALL.a.tmp: No such file or directory
```

The `.tmp` file is created through code in `verilated.mk`:

```
	$(file >$@.tmp)
	$(foreach L, $(filter-out %.a,$^), $(file >>$@.tmp, $L))
```

Running with `make --warn-undefined-variables test` results in 

```
/home/abuild/rpmbuild/BUILD/verilator-4.202/test_regress/../include/verilated.mk:210: warning: undefined variable `file >Vt_EXAMPLE__ALL.a.tmp'
/home/abuild/rpmbuild/BUILD/verilator-4.202/test_regress/../include/verilated.mk:210: warning: undefined variable `file >>Vt_EXAMPLE__ALL.a.tmp, Vt_EXAMPLE__ALL.o'
```

The [`file` make function](https://www.gnu.org/software/make/manual/html_node/File-Function.html) seems to be added in 2012 and was first released in GNU Make 4.0 (https://github.com/ndmitchell/gnumake/blob/master/ChangeLog#L234). RHEL/CentOS7 ships with GNU Make 3.82.

wsnyder, 2021-05-04:
I would encourage you to submit a pull request to fix this.  Otherwise, this will likely not be fixed, as efforts and tests are all run on newer OSes.

udif, 2021-05-04:
Just adding my 2c's:

You may manage to get Verilator to compile on RH7, but it is getting outdated, and sooner or later you will run into a required library you don't have access to.

Having to work with RH7, with no root privileges to install additional packages, my solution for building Verilator (and other packages in general) is to install Nix.
Nix is a user-level package manager, that only requires you to have a `/nix` directory owned by you (if you don't have root permission on your machine, you may need someone else to create it for you and do the correct `chown` so you have all the permissions).
Nix has a non-flat learning curve (it is a functional package manager designed to solve the dll-hell problem), but it has a huge collection of packages (if not the largest), and with a short cheatsheet you can find your way around it.

To install Nix (**not** NixOS) by:

`curl -L https://nixos.org/nix/install | sh`

(see https://nixos.org/download.html). The `install` page mentions you must have `sudo root` capabilities, but it is needed just to create the `/nix` directory. If you have that already, no root is needed.
Once Nix is installed, and you install the necessary packages, you can compile Verilator on a shell opened by:

`nix-shell -p bison flex gcc autoconf`

Once I have Verilator compiled, I can run it uner Nix, but not on a different machine, since the Verilator binaries produced relies on shared libraries stored under `/nix/...`

My solution is to copy those shared objects to the same directory where all Verilator binaries are installed, and to use patchelf to change the absolute path of these shared objects inside all the Verilator binaries:

```
TARGET=/home/udif/git/verilator/bin ; export TARGET
cp -pf `ldd verilator_bin | grep nix | sed 's/^.* => //g' | sed 's/ .*$//g'` $TARGET
patchelf --set-interpreter $TARGET/ld-linux-x86-64.so.2 --set-rpath $TARGET $TARGET/verilator*bin*
```
Now I can run those executables without any dependency on Nix.

Here is my short cheatsheet for Nix:
---
First go to bash, then in bash, run:
`. $HOME/.nix-profile/etc/profile.d/nix.sh`

Install packages by:
`nix-env -iA <package name>`

Search packages by:
`nix search <search term>`

Query available package version
don't add 'nixpkgs.' to the package name
`nix-env -qa --description packagename`

Build typical tools (verilator)
`nix-shell -p bison flex gcc autoconf`

Updates
`nix-channel --update`
`export NIXPKGS_ALLOW_UNFREE=1 # for vscode`
`nix-env -u`

Removed unused files
`nix-collect-garbage`

imphil, 2021-05-04:
I'm not using RHEL7 myself, but only build (RPM/deb) packages for it. I'll look into a fix if I find some time for it.

The wider question if RHEL7 should be supported is really a Verilator project decision -- but whatever the outcome might be, please document it as system requirement.

I know that *many* in the EDA industry are still using RHEL7 (or even RHEL6), so keeping Verilator working on those environments feels like a good thing to me.

udif, 2021-05-04:
I can only add you are correct w.r.t the status of RHEL6 & RHEL7 in the EDA community. Just to name a few:

Cadence Platform matrix:
https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/support/cadence-platform-support-plan.pdf

Synopsys Platform roadmap:
https://www.synopsys.com/support/licensing-installation-computeplatforms/compute-platforms/compute-platforms-roadmap.html

Mentor Graphics (now Siemens) Modelsim platform support:
https://static.sw.siemens-cdn.com/siemens-disw-assets/public/6gMVEbq0wlPDGdqA1yvNuE/en-US/Siemens%20SW%20HDL%20Simulation%20ModelSim%20FS%2078330%20C2.pdf

So most  EDA workstation are installed with CentOS7/RHEL7 (and some even RHEL6).

closed 2021-05-06 by wsnyder

imphil, 2021-05-07:
The builds are all green now, so at least "make test" works for all distributions with the patch from #2921. 

https://build.opensuse.org/package/show/home:phiwag:edatools/verilator-4.202

PR #2921 (2021-05-06): Add back support for gmake 3.x
The "file" make function is only available in gmake 4.x, which isn't
available on RHEL/CentOS 7. Use alternative syntax to support older
gmake versions.

Fixes #2920

We appreciate your contributing to Verilator.  If this is your first commit, please add your name to docs/CONTRIBUTORS, and read our contributing guidelines in docs/CONTRIBUTING.rst.
commit 44fd205e (2021-05-06): Fix make support for gmake 3.x (#2920) (#2921)

The "file" make function is only available in gmake 4.x, which isn't
available on RHEL/CentOS 7. Use alternative syntax to support older
gmake versions.

Fixes #2920