verilator/verilator #2924: Correct interpretation of VerilatedVcdC->dump() ???

71GA, 2021-05-05:
In my `simulation_driver.cpp` file I am simulating my input i.e. clock `i_clk` using this for loop: 

```
	for(tc = 0; tc < (1 << 22); tc++){

		tb->i_clk = 0;
		tb->eval();
		
		tf->dump(tc);
		
		tb->i_clk = 1;
		tb->eval();
		
		tc = tc + 2;
		tf->dump(tc);

	}
```

For loop in the first paragraph sets `i_clk() = 0` and because of a change I immediately evaluate the model as well. 

In the second paragraph I use `tf->dump(tc)`. This is poorly documented inside `/usr/share/verilator/include/verilated_vcd_c.h` where there is only one comment saying:

```
   /// Write one cycle of dump data
```

So I assume it should write `tc` cycle (only one cycle) of data in the `tf`.

Then the third paragraph sets `i_clk() = 1` and because of a change I immediately evaluate the model as well. 

Fourth paragraph then increments `tc` by `2` and again uses `tf->dump(tc)` that I assume will write twice as more cycles of data to the file `tf` than it did before. 

But it looks like my interpretation of `tf->dump()`is COMPLETELY wrong, because the state when `i_clk = 0` is twice as long as `i_clk = 1` and I expected it to be vice versa. Please take a look at the image.

[image]

Why is my result quite the oposite of what I expected? What is the correct interpretation of `tf->dump()`?

wsnyder, 2021-05-05:
The dump will use the time stamp at the time you call it.  Since you're calling it right after incrementing time by 2, that's what it uses.  Generally you'd call dump immediately after each eval.  Note a perhaps preferable alternative is to use the standard $dumpvar/$dumpfile routines in Verilog then you don't need to worry about calling dump yourself.

If any documentation updates would help you here, please consider making a pull request against the documentation, thanks!

closed 2021-05-05 by wsnyder

71GA, 2021-05-06:
@wsnyder  You say:
The dump will use the time stamp at the time you call it
> The dump will use the time stamp at the time you call it.

But I still don't understand **how** it uses the time stamp and **what data interval is dumped** into the file.

In my for() loop dump uses timestamps like this:

1. **itteration:** It uses `tf->dump(0)` and then ˙tf->dump(2)˙
2. **itteration:** It uses `tf->dump(3)` and then ˙tf->dump(5)˙
3. **itteration:** It uses `tf->dump(6)` and then ˙tf->dump(8)˙
4. ...

From this and the wave representation inside GTKwave I assume that `tf->dump(0)` writes the **current** i.e. **momentary** data values _(for all the model's I/O, registers...)_ as they are in the **moment of timestamp** _(in this case `0` but later `2`, `3`, `5`...)

From the documentation this is not clear because dump could write e.g. **interval** of data and from the last use until the current time stamp. 

Also in the documentation I can not find any information what unit timestamp is in. From the GTKwave I can see that it actualy represents nano seconds [ns]. I think this is some important information that is completely undocumented... Which is suprisingly because so many large companies _(NXP, ARM...)_ was supposed to use this software. I can hardly believe this is really the case because documentation is quite neglected...

71GA, 2021-05-06:
This is the correct way to use dump:

```
	tc = 0;
	while(tc < (1 << 22)){

		// G:
		tb->i_clk = 0;
		tb->eval();

		// H:
		tf->dump(tc);
		tc = tc + 1;
		tf->dump(tc);

		// I:
		tb->i_clk = 1;
		tb->eval();

		// J:
		tf->dump(tc);
		tc = tc + 3;
		tf->dump(tc);
	}
```

Paragraph `// G:` sets the input value and immediately evaluates. 
Paragraph `// H:` writes **point A** (!) at moment ˙0˙ [ns] and another **point B** (!) at moment `1` [ns].
Paragraph `// I:` sets the input value and immediately evaluates.
Paragraph `// J:` writes **point C** (!) at moment ˙1˙ [ns] and another **point D** (!) at moment `4` [ns].

**NOTE:** Pay attention that **point B** and **point C** are written at exact same moment! 
**NOTE:** `tf->dump(x)` takes an argument of `x` which is a moment in time and not an time interval! This moment is measured absolutely from the moment  `x = 0` onwards.

The waveform is now perfect _(screenshot)_! And it is easy to remember this because you just have to: _"draw all the points"_ _(**A**, **B**, **C**, **D**)_.

[image]

I think this is enough info for everyone to now understand this...