verilator/verilator #2925: Improve large lookup tables

gezalore, 2021-05-06:
I have a design which relies on large lookup tables/ROMs coded as very large Verilog constants. This eventually results in V3Premit inserting huge temporaries, and initializing them with inine code every time their use is reached:

https://github.com/verilator/verilator/blob/8624ce6a84dcd0908a9d721420e1f790ffaf4ba1/src/V3Premit.cpp#L169-L187

This results in hundreds of machine instructions being spent on constant initialzing a stack variable on every cycle, all so we can do 2 indexed lookups into the temp. This is both slow at runtime and at compile time as it bloats the function sizes in a way that output split cannot chop up.

An impovement would be to emit these constants as static data. I can have a go at this and V3Table has some helpful analogous code. What I am struggling with, is that I see all static arrays emitted in EmitC are inialized with code in _ctor_var_reset, which is still a bit wasteful and also racy (though probably harmless) with multiple simulation context. Is there a reason why we don't just emit these as C static data initializer as in:
```
static int foo[] = { 1, 2, 3,4 .... };
```
I appreciate this only works for primitive data types, so some mechanism is needed for referencig these in eg VlWide types.

gezalore, 2021-05-06:
Here is a minimum example:
```verilog
module top(
   input   wire  [7:0] i,
   output  wire  [7:0] o
 );

   wire [255:0][7:0] C = {8{256'h426347602346043448640821046421423442144443251352146532562367247}};

   assign o = C[i];
endmodule
```

Which yields:
```c
VL_INLINE_OPT void Vwide::_combo__TOP__1(Vwide__Syms* __restrict vlSymsp) {
   VL_DEBUG_IF(VL_DBG_MSGF("+    Vwide::_combo__TOP__1\n"); );
   Vwide* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
   // Variables
   VlWide<64>/*2047:0*/ __Vtemp1;
   // Body
   __Vtemp1[0U] = 0x62367247U;
   ...
   __Vtemp1[0x3fU] = 0x4263476U;
   vlTOPp->o = (0xffU & (((0U == (0x1fU &`
                                  ((IData)(vlTOPp->i)`
                                   << 3U)))
                           ? 0U : (__Vtemp1[
                                   ((IData)(1U)`
                                    + (0x3fU`
                                       & ((IData)(vlTOPp->i)`
                                          >> 2U)))]`
                                   << ((IData)(0x20U)`
                                       - (0x1fU`
                                          & ((IData)(vlTOPp->i)`
                                             << 3U)))))`
                         | (__Vtemp1[(0x3fU`
                                      & ((IData)(vlTOPp->i)`
                                         >> 2U))]`
                            >> (0x1fU & ((IData)(vlTOPp->i)`
                                         << 3U)))));
}
```

wsnyder, 2021-05-06:
Currently the init is normal code because as you mention there could be complex equations or function calls or otherwise to set the values.  Some possible optimizations:
1. Convert all-constant inits to initializer lists, as you suggest.  Ideally this would be transformed before V3EmitC. V3EmitC is doing too much and should ideally be closer to just a dump-the-tree formatter.
2. Convert temporaries that have constant values to class const statics like V3Table.  The statics would then be inited in the single startup (either using the optimization above, or the longer code).  I worry this might result in slower code though as the compiler won't be able to constant propagate.  So perhaps it needs some size threshold (e.g. do it for 32 bytes or more or something).

closed 2021-05-10 by gezalore