verilator/verilator #2936: multiplication (signed * unsigned)

jeras, 2021-05-08:
I am trying to implement RISC-V (RV32IM) 32-bit signed * unsigned multiplication with simple non optimized code:
```SystemVerilog
mul [64-1:0] = $signed(a[32-1:0]) * $unsigned(b[32-1:0]);
```
I expected it to work, but the RISC-V architecture tests fail on the next example: 
https://github.com/jeras/rp32/tree/master/issues/mul

```
a = -32'h20000001;
b = -32'h20000001;
```
If I perform the calculation with literals I get the right result:
```
a = 0xdfffffff =  -536870913, b = 0xdfffffff = 3758096383
r64ws = 0xe3ffffff40000001 = -2017612636283207679
r64ds = 0xe3ffffff40000001 = -2017612636283207679
```
But if I use signals `$signed(a) * $unsigned(b)` I get a wrong result:
```
signed/unsigned multiplication:
r64ws = 0xc3fffffe40000001 = -4323455649791868927
r64ds = 0xc3fffffe40000001 = -4323455649791868927
```

I tried to simulate in Rivera-PRO on [edaplayground](https://www.edaplayground.com/x/dyaU), but I get the same result as with Verilator.
I also tried with smaller vectors, in case the issue is related to 32-bit integer types, but I get the same behavior.

Is there something I am doing wrong, is this a bug in both tools, ...

The SystemVerilog LRM does not seem to provide any relevant details here, I only found section 11.4.3.1 to be somehow relevant.

Verilator here reports a width warning, again I am not sure what should be the width of a multiplication result according to the LRM.

wsnyder, 2021-05-08:
The IEEE rules are both sides must be signed to get signed math, so you're doing unsigned multiplies.
The IEEE rules say the size will be determined by the output which is 64 bits, although your arguments are 32 bits (which is why it's a WIDTH warning.
So if a is signed it will be used as a large integer, and the $unsigned will be on 64-bits not 32-bits, hence the big result.
I suspect this isn't what you wanted, make a temporary variable of the correct width, or a cast, to better indicate intent.

(It's also possible there's a bug here, but until another simulator disagrees... ;)

closed 2021-05-08 by wsnyder

jeras, 2021-05-08:
I tried with the other simulators and got the same results.
I will have to figure how to do it right.

I tried to extend both variables (signed and unsigned) to 64-bit and now I get the right result.
Only sign extending the first operand also seems to work.
I am a bit preoccupied how this will affect synthesis.