verilator/verilator #2938: Got incorrect warning message in latest releases version 4.202

yanjiuntw, 2021-05-10:
Hello,

I use latest releases version to check lint of my verilog file, but I got a incorrect message.

> What 'verilator --version' are you using?  Did you try it with the git master version?
```text
$ verilator --version
Verilator 4.202 2021-04-24 rev v4.202
```
> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Here is my verilog file content: 
```verilog
// test.v
module test (
  input      [2:0] a,
  input      [7:0] c,

  output reg [7:0] b
);

  integer i;

  always @ (*)
  begin
    case(a)
      {3'b000}:
        b = 8'd1;
      {3'b001}:
        for(i=0;i<4;i=i+1) b[i*2+:2] = {2{c[i]}};
      {3'b010}:
        b = 8'd3;
      {3'b011}:
        b = 8'd4;
      default: b = 0;
    endcase
  end

endmodule
```
And I use this command run the test, and got a Warning-LATCH message:
```text
$ verilator --lint-only -Wall test.v
%Warning-LATCH: test.v:10:3: Latch inferred for signal 'b' (not all control paths of combinational always assign a value)
                           : ... Suggest use of always_latch for intentional latches
   10 |   always @ (*)
      |   ^~~~~~
                ... For warning description see https://verilator.org/warn/LATCH?v=4.202
                ... Use "/* verilator lint_off LATCH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

> Would you be willing to try to fix Verilator yourself with assistance?

I will try if I can understand what wrong.

wsnyder, 2021-05-10:
Makes sense.  I suspect it gets confused by the bit select range. @margej mind taking and look and making a proposal to fix it?

margej, 2021-05-11:
Latch detection graph does not detect assignment to 'b' in the 3rd if (as commented below). This is probably because the assignments to b here depend upon the current value of b:

            // Body
            if ((4U & (IData)(vlTOPp->a))) {
                vlTOPp->b = 0U;
            } else {
                if ((2U & (IData)(vlTOPp->a))) {
                    vlTOPp->b = ((1U & (IData)(vlTOPp->a))
                                  ? 4U : 3U);
                } else {
                    if ((1U & (IData)(vlTOPp->a))) { // Assignment to b NOT detected here
                        vlTOPp->b = ((0xf0U & (IData)(vlTOPp->b)) 
                                     | ((4U & ((IData)(vlTOPp->c) 
                                               << 1U)) 
                                        | (1U & (IData)(vlTOPp->c))));
                        vlTOPp->b = ((0xfU & (IData)(vlTOPp->b)) 
                                     | ((0x40U & ((IData)(vlTOPp->c) 
                                                  << 3U)) 
                                        | (0x10U & 
                                           ((IData)(vlTOPp->c) 
                                            << 2U))));
                    } else {
                        vlTOPp->b = 1U;
                    }
                }
            }

wsnyder, 2021-05-11:
Can you please convert that to a pull, along with a test, then we can merge it.  Thanks for debugging etc!

margej, 2021-05-11:
Sure, I'll add the test case provided by @yanjiuntw above (when I remind myself how to add a test)

margej, 2021-05-12:
I just realised I've been posting all my comments on the PR, not this issue, so 1, the above was resolved by #2921 
2. Please see https://github.com/verilator/verilator/pull/2948#issuecomment-839942391 for question on resolving new test failure

margej, 2021-05-28:
Re: Issues on PR #2948 I need to distinguish AstIf nodes used for array bounds checking from those present in the RTL. It _appears_ these are generated in V3Unknown.

My hope is I can flag them at creation so they can be later ignored for the purposes of latch detection

wsnyder, 2021-06-01:
Thanks for your work on this.

closed 2021-06-01 by wsnyder