verilator/verilator #2949: Is final statement supported for 4.200 ?

agrobman, 2021-05-11:
How may we help - what is your question?

wsnyder, 2021-05-11:
Verilator uses the standard simulator -y and +incdir+ to find the directories of verilog files.  Hopefully you don't have files of the same name in multiple directories, as that's not defined by any standard (AFAIK).

agrobman, 2021-05-11:
our compilation command was:

verilator  -f flist --top-module tb_top -exe test_tb_top.cpp

and content of flist was:
```
+libext+.v+.sv
+incdir+$RV_ROOT/verif/vtb
common_defines.vh
ehx3_pdef.vh
-f $RV_ROOT/design/flist // design files
$RV_ROOT/verif/core/dpi/dpi_imports_verilator.svh
$RV_ROOT/verif/vtb/tb_top.sv
$RV_ROOT/verif/vtb/tracer.sv
-v $RV_ROOT/verif/vtb/event_counter.sv
$RV_ROOT/verif/vtb/tilelink/tl_trans.sv
$RV_ROOT/verif/vtb/tilelink/tl_slv.sv
$RV_ROOT/testbench/remote_bitbang.cc
$RV_ROOT/testbench/SimJTAG.cc
$RV_ROOT/testbench/SimJTAG.v
$RV_ROOT/verif/core/dpi/whisper-client.cpp
gitstate.v

```
verilator was taking test_tb_top.cpp from
*/verif/testbench

I was thinking that +incdir+ is used for `include files search only ...

?

wsnyder, 2021-05-11:
Currently -y and +incdir+ are treated the same.