verilator/verilator #2950: packed port support

lucasbrasilino, 2021-05-11:
Hello!

Out of curiosity, are there plans to support packed module ports ? 

Thanks!

lucasbrasilino, 2021-05-11:
Hello,

I'm referring to section `23.2.2.2 ANSI style list of port declarations`, that details `list_of_port_declarations` in section `A.1.3`, where we can find `net_port_header` , which following down all definitions we get to A.2.2.1:
```
implicit_data_type ::= [ signing ] { packed_dimension }
```
As example::
```
module foo (
  input  wire [7:0][31:0] d_in,
  output wire [7:0][31:0] d_out
)
```
Thanks!

lucasbrasilino, 2021-05-15:
Since what version is it supported ? 
I use verilator as my cocotb backend simulator. Currently running version 4.106 due to #2778.  
Attached goes the module I'm playing with. 

[foo.zip](https://github.com/verilator/verilator/files/6488394/foo.zip)

wsnyder, 2021-05-16:
This works on the current master, I didn't track down which commit did that - feel free to bisect if that's interesting to you.

closed 2021-05-16 by wsnyder