verilator/verilator #2952: ../V3OptionParser.h:142:26: error: invalid initialization

gkamendje, 2021-05-12:
Using the latest version on github 
```
commit e3b20a3ea228f46d9178cb71dfc1ffd7880cd7db
Author: Yutetsu TAKATSUKASA <y.takatsukasa@gmail.com>
Date:   Wed May 12 00:45:43 2021 +0900
```
I  get a couple of invalid initialization errors and the compilation fails. I do not see these errors with previous version (4.030 and 4.202) 
```/usr/bin/flex -d -oV3PreLex_pregen.yy.cpp ../V3PreLex.l
In file included from ../V3OptionParser.cpp:22:0:
../V3OptionParser.h: In constructor âV3OptionParser::AppendHelper::AppendHelper(V3OptionParser&)â:
../V3OptionParser.h:142:26: error: invalid initialization of non-const reference of type âV3OptionParser&â from an rvalue of type â<brace-enclosed initializer list>â
         : m_parser{parser} {}
                          ^
/usr/bin/python3 ../flexfix V3PreLex <V3PreLex_pregen.yy.cpp >V3PreLex.yy.cpp
/usr/bin/python3 ../flexfix V3Lexer <V3Lexer_pregen.yy.cpp >V3Lexer.yy.cpp
In file included from ../V3OptionParser.cpp:22:0:
../V3OptionParser.h: In constructor âV3OptionParser::AppendHelper::AppendHelper(V3OptionParser&)â:
../V3OptionParser.h:142:26: error: invalid initialization of non-const reference of type âV3OptionParser&â from an rvalue of type â<brace-enclosed initializer list>â
         : m_parser{parser} {}
                          ^
g++ -std=gnu++11  -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3PreProc.cpp -o V3PreProc.o
g++ -std=gnu++11  -MMD -I. -I.. -I.. -I../../include -I../../include -O2 -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3PreProc.cpp -o V3PreProc.o
g++ -std=gnu++11  -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseLex.cpp -o V3ParseLex.o
make[2]: *** [V3OptionParser.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [V3OptionParser.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ../V3Options.cpp:24:0:
../V3OptionParser.h: In constructor âV3OptionParser::AppendHelper::AppendHelper(V3OptionParser&)â:
../V3OptionParser.h:142:26: error: invalid initialization of non-const reference of type âV3OptionParser&â from an rvalue of type â<brace-enclosed initializer list>â
         : m_parser{parser} {}
                          ^
In file included from ../V3Options.cpp:24:0:
../V3OptionParser.h: In constructor âV3OptionParser::AppendHelper::AppendHelper(V3OptionParser&)â:
../V3OptionParser.h:142:26: error: invalid initialization of non-const reference of type âV3OptionParser&â from an rvalue of type â<brace-enclosed initializer list>â
         : m_parser{parser} {}
                          ^
make[2]: *** [V3Options.o] Error 1
make[2]: *** [V3Options.o] Error 1
make[2]: Leaving directory `/data/stealth_2/g243135/verilator/src/obj_dbg'
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make[2]: Leaving directory `/data/stealth_2/g243135/verilator/src/obj_opt'
make[1]: *** [../bin/verilator_bin] Error 2
make[1]: Leaving directory `/data/stealth_2/g243135/verilator/src'
make: *** [verilator_exe] Error 2
System time: 175.208 User time: 1604.728 Elapsed Time: 1:01.90
```
I am running 
```
bison --version
bison (GNU Bison) 3.0.4
```
Thanks for any hint. 
G

yTakatsukasa, 2021-05-12:
The latest master can be built on CentOS7.
Thank you for reporting.

closed 2021-05-12 by yTakatsukasa

gkamendje, 2021-05-13:
Thanks for fixing. 
G

PR #2954 (2021-05-12): Internals: Fix build failure on older gcc such as 4.8.5 on CentOS7
Fixes #2952

I will merge after CI passes.