verilator/verilator #2953: V*__ALL.a archive creation error in version 4.202

gkamendje, 2021-05-12:
Using `Verilator 4.202 2021-04-24 rev v4.202` in conjunction with https://github.com/chipsalliance/Cores-SweRV   81bff1c0d73efb5ab4ba66461e34430c87cebfac, verilator fails to properly create the archive `Vtb_top__ALL.a`. I get the following error. 
```
Archive ar -rcs Vtb_top__ALL.a Vtb_top.o Vtb_top__1.o Vtb_top__2.o Vtb_top__3.o Vtb_top__4.o Vtb_top__5.o Vtb_top__6.o Vtb_top__7.o Vtb_top__8.o Vtb_top__9.o Vtb_top___024unit.o Vtb_top__Slow.o Vtb_top__1__Slow.o Vtb_top__2__Slow.o Vtb_top__3__Slow.o Vtb_top__4__Slow.o Vtb_top__5__Slow.o Vtb_top__6__Slow.o Vtb_top__7__Slow.o Vtb_top__8__Slow.o Vtb_top___024unit__Slow.o Vtb_top__Syms.o
ar: @Vtb_top__ALL.a.tmp: No such file or directory
g++    test_tb_top.o verilated.o Vtb_top__ALL.a      -o Vtb_top
g++: error: Vtb_top__ALL.a: No such file or directory
make[1]: *** [Vtb_top] Error 1
```
I do not get this error message when running with `verilator 4.030`. I have tried modifying the `OPT_FAST` option but it does not help. 
After running  `verilator --cc *`  which does not produce any error, the last verilator step is invoked with the following command.
`make -j  -C obj_dir/ -f Vtb_top.mk OPT_FAST="-Os"`
Manually entering the archive file creation command on the command prompt does not produce any error message. 
` $ ar -rcs Vtb_top__ALL.a Vtb_top.o Vtb_top__1.o Vtb_top__2.o Vtb_top__3.o Vtb_top__4.o Vtb_top__5.o Vtb_top__6.o Vtb_top__7.o Vtb_top__8.o Vtb_top__9.o Vtb_top___024unit.o Vtb_top__Slow.o Vtb_top__1__Slow.o Vtb_top__2__Slow.o Vtb_top__3__Slow.o Vtb_top__4__Slow.o Vtb_top__5__Slow.o Vtb_top__6__Slow.o Vtb_top__7__Slow.o Vtb_top__8__Slow.o Vtb_top___024unit__Slow.o Vtb_top__Syms.o`

I suspected that there could be a race in the parallel call to make so I tried removing the "-j" option from the call to make without success. 

The generated file` obj_dir/Vtb_top.mk` does not contain any explicit rule to create the archive file `Vtb_top__ALL.a`. I guess  Verilator uses either some implicit rules or has a set of internal rules that it uses for this goal. Any clarification around this point would be very instructive for me. 

Unfortunately, I do not know if this issue exists with the latest git version given that compilation of the latest git version does not work for me. See issue https://github.com/verilator/verilator/issues/2952

Would you be willing to try to fix Verilator yourself with assistance?
Yes I would try and fix Verilator myself if someone could provide some assistance.
Thanks for any hint.
G

wsnyder, 2021-05-13:
With #2952 now fixed can you confirm if this now works ok (which I suspect/hope)?  Thanks

gkamendje, 2021-05-13:
Confirming that I do not see the issue after the fix proposed in #2952.

closed 2021-05-13 by gkamendje