verilator/verilator #2955: make test fails %Error: Cannot open -f command file: input.vc

gkamendje, 2021-05-13:
Using the latest git master, running make test fails with the following errors:
```
Starting Verilator 4.203 devel rev v4.202-28-ge3b20a3
Starting Verilator 4.203 devel rev v4.202-28-ge3b20a3
%Error: Cannot open -f command file: input.vc
%Error: Exiting due to 1 error(s)
[Inferior 1 (process 36473) exited with code 01]
No stack.
%Error: export VERILATOR_ROOT=/ver_trial/verilator
%Error: gdb /ver_trial/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc -OD --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc t/t_a1_first_cc.v" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed gdb /ver_trial/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc -OD --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc t/t_a1_first_cc.v" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Warning: vlt/t_a1_first_cc: Exec of perl failed: warning: File "/ver_trial/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load:/usr/bin/mono-gdb.py:/usr/lib/golang/src/runtime/runtime-gdb.py".

vlt/t_a1_first_cc: %Error: Exec of perl failed: warning: File "/ver_trial/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load:/usr/bin/mono-gdb.py:/usr/lib/golang/src/runtime/runtime-gdb.py".
vlt/t_a1_first_cc: FAILED: Exec of perl failed: warning: File "/ver_trial/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load:/usr/bin/mono-gdb.py:/usr/lib/golang/src/runtime/runtime-gdb.py".
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:09
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:09

======================================================================
        #vlt/t_a1_first_cc: %Error: Exec of perl failed: warning: File "/ver_trial/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load:/usr/bin/mono-gdb.py:/usr/lib/golang/src/runtime/runtime-gdb.py".
                make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:09
make: *** [smoke-test] Error 10
```
I see the same issue with version 4.202 and 4.030. 

```
Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_ROOT     = /ver_trial/verilator
    SystemC system-wide = 0

Environment:
    MAKE               = make
    PERL               =
    SYSTEMC            =
    SYSTEMC_ARCH       =
    SYSTEMC_INCLUDE    =
    SYSTEMC_LIBDIR     =
    VERILATOR_ROOT     = /ver_trial/verilator
    VERILATOR_BIN      =
```
I am running CentOS7 with 
`gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)`

Thanks for any hint. 
G

wsnyder, 2021-05-13:
Nothing recent I can think of would affect this.  Are you sure input.vc exists as it should (wasn't accidentally deleted)?

gkamendje, 2021-05-13:
I do see the file ${VERILATOR_ROOT}/test_regress/input.vc with the following content 
```
+librescan +notimingchecks +libext+.v
 -y t
 +incdir+t
 +incdir+../include
```
Where is file supposed to the located?

wsnyder, 2021-05-13:
That is the right spot, perhaps the earlier change directory was to the wrong place?  Anyhow if you could debug I suspect it is something local to you as there are many centos users, if not a pull request would be appreciated.

gkamendje, 2021-05-13:
Hmm.. not sure which "change directory" you are referring too. Anyway, my observations are the following:
1) The problem is related to smoke-test. 
2) Looking at the Makefile, `smoke-test`launches `test_regress/t/t_a1_first_cc.pl` from `${VERILATOR_ROOT}` which implies that the script expects the file`input.vc` to be located directly under `${VERILATOR_ROOT}`. However, what I can see is that `input.vc` is located under `test_regress`. As can be seen below, during smoke-test execution, there is no `cd test_regress` after the last exit from the src directory and the call to `test_regress/t/t_a1_first_cc.pl`. It is therefore not surprising that the script cannot find `input.vc`. 
```
make[1]: Leaving directory `/ver_trial/verilator/src'
test_regress/t/t_a1_first_cc.pl
```
This happens even with a fresh clone of the git master. Other users on my system are seeing the same behavior. 
I suspect that it is something related to our system.

imphil, 2021-05-20:
To debug this further can you run 

```
make --warn-undefined-variables SHELL='sh -x' test
```

from the directory where you checked out/built Verilator and attach the full log output to this issue?

wsnyder, 2021-06-06:
I'm assuming you figured this out, if not please feel free to post additional questions and reopen.

closed 2021-06-06 by wsnyder