verilator/verilator #2957: Recursive function fails to compile

treforsouthwell, 2021-05-13:
The code below builds on Cadence Xcelium but fails in Verilator:

/verilator --version
Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)

./verilator --cc test2.sv
%Error: test2.sv:6:16: Found definition of 'log2_f' as a VAR but expected a task/function
    6 |         return log2_f(a/2) + 1;
      |                ^~~~~~
%Error: Exiting due to 1 error(s)

-- test2.sv --
module fred;
   function automatic int log2_f(int a);
      if (a == 1)
        return 0;
      else
        return log2_f(a/2) + 1;
   endfunction // log2_f
endmodule

wsnyder, 2021-05-13:
1. This code will break if a=0.
2. Don't use a recursive algorithm when it isn't needed. A for loop is more obvious and faster, but even better you can just use $clog and some math.
3. Verilator doesn't currently support recursive functions. Agree this should be fixed, and until then at least a more obvious UNSUPPORTED error message.

treforsouthwell, 2021-05-13:
Thanks for your reply, this is pre-existing code taken from a database. I'm really trying to see if our code can build with Verilator.
So yes, maybe the ticket is then a feature request to support recursion.
Thanks

closed 2021-06-13 by wsnyder

kaddkaka, 2024-03-28:
Does verilator support recursive functions calls yet? Is there a plan to supoprt it?

wsnyder, 2024-03-28:
Recursive is supported.