verilator/verilator #2958: Can we make verilated code more ccache-friendly?

rswarbrick, 2021-05-14:
In the OpenTitan project, we use Verilator to (among other things) build a chip-level simulation. This is now reasonably large: it takes a minute or two to verilate and then, if nothing is cached, something like 60-90mins CPU time to compile the resulting C++ code. Using ccache improves things, but a small change to the design tends to result in all the C++ code changing slightly ("shuffling along" past split boundaries).

It turns out that backup software has a similar problem. There, they chunk up big files into (say) ~1MiB blocks to upload. But they don't want to have to upload everything again if you insert a byte at the start of the file. For backup software, there seems to be a standard(ish) solution to the problem. Instead of you splitting every 1MiB, you define a windowed hash function that you compute as you read through the file. Every time the function evaluates to zero, you split. For roughly 1MiB blocks, you'd want a hash that's evenly distributed on `[0, 2^20]`. The trick is that now the split positions depend on the local content, rather than the absolute position in the file, and you solve the "shuffling past boundaries" problem. A much lower-tech example of the same idea is when people write documentation with a sentence to a line, rather than reflowing, to avoid big diffs when a sentence changes length. (Here, the analogue of a block with hash zero is "the end of a sentence").

For some more description about this scheme, see [this 2015 blog post](https://restic.net/blog/2015-09-12/restic-foundation1-cdc/) describing the algorithm used by [Restic](https://restic.net/).

Anyway, I think we could do the same with verilator. When assigning statements to output (C++) functions, we'd use their hashes (which I believe are already defined), modulo some configurable value, to decide where to split. We'd also presumably have to either use one function per output file or do this again when assigning functions to files. We'd also need to change the output function naming scheme, to make sure everything is stable.

The idea is that now if I make a small tweak to the Verilog design, I might expect to perturb just 1 or 2 files, instead of half the design (everything to the "right" of where my tweak happens to fall).

I've just chatted with @gezalore as a sanity check. I think he might be interested on working on this. @wsnyder: Does this sound like a sane idea to you?

wsnyder, 2021-05-14:
#2140 has some similar discussions, I suspect the header modifications there are a critical first step for what you propose to work well.

I'd certainly welcome effort on these.

gezalore, 2021-05-15:
@wsnyder the current V3Hash/V3Hashed implementation relies on pointer values, so it's not stable across multiple invocations of Verilator. For this scheme to work, we need a tree hash which is identical across invocations over the same design. Would you prefer me changing the V3Hashed scheme to be stable, or to design a new visitor/add a new virtual method to every node? Do you have any better suggestions?

With the current V3Hashed, there might be issues trying to do what we need as the sameHash would have to call back into V3Hashed to compute stable hashes of non-child but relevant member nodes.

wsnyder, 2021-05-16:
It would be nice if V3Hashed could be made stable, perhaps you could give it a try?  My concern thinking about it quickly is that I'm not sure if performance will be an issue.

gezalore, 2021-05-16:
@rswarbrick another slight problem is that as opposed to the file backup use case of determining split points via running hashes where each input item in the hash window is the same size (a byte or word), in Verilators, each input is rougly one top level statement (e.g.: an `assign` or a statement direcly under an `always` block), the size of these then of course varies. If you have any more algorithmic ideas on how to address this, please let me know.

rswarbrick, 2021-05-16:
Yeah, indeed. But isn't the splitting done in terms of the number of these top-level statements already? I *think* that the proposed changes don't make that any worse (or better, sadly).

gezalore, 2021-05-16:
My wory is that either you will end up with very large output files, if you just use the hash (as the likelyhood of N small trees hashing to the same value as N very large trees is hopefully the same, there is a good chance you will bundle a lot of large statements into the same output file), or you will be back with the same problem of statements shuffling if you split based on function size as well (--output-split-cfunc). Maybe the solution is to split both based on size, and running hash, which means all functions within a hashed block might still need to rebuild if things suffled due to a size split within a block?

gezalore, 2021-05-16:
@wsnyder the more interesting statement combining bits in V3Combine seem to be ifdefed out. Is there an intention to resurrect this at any point? I'm asking because the current V3Hash `depth()` method and the fact that the hash keeps track of the tree depth only seems to be used by the disabled code. I wonder how much of that functionality needs to be preserved.

wsnyder, 2021-05-16:
I don't expect to do more work on combining statements, feel free to make a standalone pull that removes these.

wsnyder, 2021-05-16:
As to hashing and preserving, perhaps an easier (?) solution is to have Verilator create a data file as to the previous run's split points, then use this to seed the new run. That is, if it sees the previous split when it's between 75%-125% of the optimal split point it will instead split at the previous split.  Otherwise it will split at the 125%.

gezalore, 2021-05-16:
I might be missing something, but I don't see it. How do I reliably identify the split point from one run to another (i.e.: what do I emit on the first run and how do I compare against it)?

wsnyder, 2021-05-16:
If no split data file (e.g. first run):
* Computing hashes of statement split points
* Split if statement_count >= split_limit (same as current)
* Write file with hash that started each hit point

If split file:
* Read file
* Computing hashes of statement split points
* Split if (statement_count >= 125% * split_limit || (statement_count >= 75% && hash in-list-of remembered_split_points))
* Write file with hash that started each hit point

gezalore, 2021-05-16:
Ok, I'm with you I thought you had some scheme that doesn't require a stable hash. But what you are saying is that instead of the running window hash, just remember the hashes of where we split on the first run, and split at the same points if near the limit, which makes more sense.

rswarbrick, 2021-05-19:
The only problem with this "first run / later run" distinction is that it would be really nice if this flow worked well:
```
git checkout branch1
make
git checkout branch2
make
git checkout branch1
hack hack hack
make
```
The nice thing about ccache is that it doesn't care about the current state of your build tree: I think that the proposal above would care, which would break this sort of flow.

Wouldn't it be easier just to operate on windows of (say) 16 statements to get more hash variance and tolerate more variation in output size? (Maybe with a hard lower limit of 10% of requested size and upper limit of 200% for degenerate cases)

gezalore, 2021-05-19:
The problem isn't really the hash variance. It's the granularity of the statements. For each statement you get one hash, but the size of a statement can vary from a simple assignment to an if statement with 10000 child statements (maybe that's my fault, but Alogic generates things like that). You really really don't want to put large statements together in the same function, that is what causes compilation time blowup due to O(n^2) algorithms in compiler. This means you need to split functions when things get big. The scenario I was worried about was that using a rolling hash would not split sequences of these large things often enough (because you still only get one hash for a huge tree, so the likelyhood of N small trees hashing to 0 is the same as N large trees hasing to 0), but I think it would be ok to just split based on size and hash, and any files between two hash splits would then have to be rebuilt, which I think is actually almost the same algorithm, but the inteval between two zero hashes is subdivided by size.

Regarding the proposed workflow: I think some data would have to persist between runs to solve the problems with the symbol table/headers described in #2140, but ideas would be great if you have any.

rswarbrick, 2021-05-19:
Oh, I see. Could we do something like this Python-ish pseudocode?
```python3
window = MyStmtWindow()
cur_size = 0
for stmt in stmts:
  cur_size += stmt.size()
  window.take_stmt(stmt)
  hash_mod = int(100 * exp_fun_size / (100 + cur_size))
  if (window.hash() % hash_mod) == 0:
    split_here()
    window.clear()
    cur_size = 0
```

The idea is that we split whenever the hash is zero, modulo `hash_mod`. If we weren't being clever, we'd want `hash_mod` to equal the requested number of statements per chunk (which I called `exp_fun_size` above). To be cleverer, we make it smaller as `cur_size` gets bigger. I've been really lame and used the "100" as a sensitivity parameter. With a bit of thought, we could probably come up with a better function that actually gives the right expected segment length.

rswarbrick, 2021-05-19:
For the symbol table question: that's a good point. A rather lame approach that I can think of uses more hashes(!). Suppose there are (say) 900 fields needed of a given type. Round this up to maybe 1024 or 2048 and hash each field name, using some sort of Cuckoo hashing scheme to ensure everything's distinct. Then store an array of those fields and index them in the generated C++ code by fixed offsets.

Then the header will be stable for small changes in field counts or field names. If there's a change that hits a hash collision, we'll have to recompile those files but that shouldn't happen too often.

The advantage is that you get quick (hopefully) rebuilds which are reasonably likely to cache across changes. The disadvantage is that you increase total memory size because of oversizing the array, and probably harm dcache locality a bit.

wsnyder, 2021-05-19:
Changing the order of signals depending on model can be around 20% performance. If we can find one I'd prefer a scheme that performs as well as "normal", even if somewhat more recompiles.  FWIW I suspect (without data) that add/removes of a signal or two are more common than renames.

rswarbrick, 2021-05-19:
> Changing the order of signals depending on model can be around 20% performance.

Yikes! Hmm, maybe the hashing scheme isn't such a good idea...

rswarbrick, 2021-05-19:
I guess the other thing that we could do would be to mirror the chunking we're describing for the statements. So we'd have (say) blocks of 32 variables that we aim to be approximately 50% full. We'd fill them in order, but jump to the next block if a hash was zero. Otherwise the scheme would be as above. This would maintain the ordering as above, but hopefully would solve the "shuffling" problem on a single insertion or removal.

gezalore, 2021-07-22:
With #3071, the single threaded model (without tracing) on OpenTitan gets about 95% ccache hit rate when changing small amounts of logic, or about 35% hit rate when renaming/adding/removing a small number of signals. This is without relying on the hash based splitting as there is enough independence between the non-inlined modules to make this work as is, but this means that anything with --flatten will still rebuild. Trace functions still need review and might cause rebuilds. Unfortunately the partitioning algorithm for multi-threaded models can generate very different results for small input changes, so not much improvement there yet..

tomager, 2024-04-16:
Improvements on this topic and incremental verilating generally would be really nice to see.