verilator/verilator #2968: Future of Verilator Parser?

udif, 2021-05-18:
@wsnyder , I've recently been made aware of https://github.com/antmicro/verilator that uses Surelog as a parser frontend.
I was wondering how much of this effort is coordinated with you.
Do you see this effort replacing the current parser?
Do you know if it solves existing problems in the Verilator parsing and elaboration phases, such as recursive functions and modules, and elaboration of false generate/endgenerate paths?

wsnyder, 2021-05-18:
Yes, I'm aware of the ongoing Surelog effort.  The hope is in the long term it can replace the parser and solve the issues you mention, but as it is not yet doing what the current parser can it's premature to say what issues exactly will be resolved.

wsnyder, 2021-05-18:
Note also the current Verilator parser is still being maintained/improved as needed, the only impact on the current code is the current parser is unlikely to get completely rewritten into something more capable.

closed 2021-05-18 by wsnyder