verilator/verilator #2969: UNOPTFLAT lint warning (for tree arbiter)

rswarbrick, 2021-05-18:
If you download the attached `prim_arbiter_tree.sv` and run Verilator lint on it, you see the following warnings:
```
> verilator --lint-only -Wno-ALWCOMBORDER tst/prim_arbiter_tree.sv
%Warning-UNOPTFLAT: tst/prim_arbiter_tree.sv:45:39: Signal unoptimizable: Feedback to clock or circular logic: 'prim_arbiter_tree.gen_normal_case.idx_tree[5:4]'
   45 |     logic [2**(IdxW+1)-2:0][IdxW-1:0] idx_tree /*verilator split_var*/ ;
      |                                       ^~~~~~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=4.203
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    tst/prim_arbiter_tree.sv:45:39:      Example path: prim_arbiter_tree.gen_normal_case.idx_tree[5:4]
                    tst/prim_arbiter_tree.sv:108:11:      Example path: ALWAYS
                    tst/prim_arbiter_tree.sv:43:39:      Example path: prim_arbiter_tree.gen_normal_case.sel_tree[2]
                    tst/prim_arbiter_tree.sv:108:11:      Example path: ALWAYS
                    tst/prim_arbiter_tree.sv:45:39:      Example path: prim_arbiter_tree.gen_normal_case.idx_tree[5:4]
%Warning-UNOPTFLAT: tst/prim_arbiter_tree.sv:46:39: Signal unoptimizable: Feedback to clock or circular logic: 'prim_arbiter_tree.gen_normal_case.data_tree[2]'
   46 |     logic [2**(IdxW+1)-2:0][DW-1:0]   data_tree /*verilator split_var*/ ;
      |                                       ^~~~~~~~~
                    tst/prim_arbiter_tree.sv:46:39:      Example path: prim_arbiter_tree.gen_normal_case.data_tree[2]
                    tst/prim_arbiter_tree.sv:108:11:      Example path: ALWAYS
                    tst/prim_arbiter_tree.sv:43:39:      Example path: prim_arbiter_tree.gen_normal_case.sel_tree[2]
                    tst/prim_arbiter_tree.sv:108:11:      Example path: ALWAYS
                    tst/prim_arbiter_tree.sv:46:39:      Example path: prim_arbiter_tree.gen_normal_case.data_tree[2]
%Error: Exiting due to 2 warning(s)
```

This is with the current head of master:
```
> verilator --version
Verilator 4.203 devel rev v4.202-41-g0f7ec6c9
```

The code is a (slightly hacked) version of a tree arbiter [from OpenTitan](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/rtl/prim_arbiter_tree.sv). This is structured as a full binary tree and I've set parameters so that it has 4 leaves, giving the following indices:
```
     0
    / \
   /   \
  1     2
 / \   / \
|   | |   |
3   4 5   6
```

I chose parameters to give a data width of just 1 bit, which makes the apparent mutual dependency between `data_tree` and `sel_tree` as easy as possible to debug. I can't see any path between them, so I *think* this is a bug?

[prim_arbiter_tree.sv.txt](https://github.com/verilator/verilator/files/6502367/prim_arbiter_tree.sv.txt)

rswarbrick, 2021-05-20:
Ahah. Now I understand what's going on. The problem is that the `split_var` hint works but there's a big `always_comb` block that also needs splitting. I'm going to push a patch on the OpenTitan side that splits the `always_comb` block.

Oddly, it looks like judicious use of `isolate_assignments` should be able to get this working, but I haven't had any luck. In particular, manually splitting out the definition of `sel` like this:
```systemverilog
          always_comb begin : p_node
            // forward path (requests and data)
            // each node looks at its two children, and selects the one with higher priority
            sel = ~req_tree[C0] | ~prio_tree[C0] & prio_tree[C1];
          end
          always_comb begin : p_node2
            // propagate requests
...
```
seems to break the cycle, but leaving the `always_comb` block intact and marking `sel` with `isolate_assignments` doesn't seem to help. Am I missing something?

wsnyder, 2021-06-04:
Not sure why isolate_assignments didn't work, needs some debugging, if you'd like to do this it would be appreciated, otherwise adding it to the list to work on.

yTakatsukasa, 2022-02-26:
When I tried the latest master, `isolate_assignments` works, but does not help resolving `UNOPTFLAT` simply because V3SplitAs comes after V3Split.
https://github.com/verilator/verilator/blob/e52a4ac74f343d774b0202335b82552a4f93496d/src/Verilator.cpp#L330-L332

If I move V3SplitAs before V3Split, `UNOPTFLAT` goes away.

I have a draft branch that can split always_comb for `sel` without any metacomment. I will send a PR after #3317.