verilator/verilator #2970: Regression in lint check ("Wide Op w/ no temp" internal error)

rswarbrick, 2021-05-18:
Commit 4f36e3e causes a regression when linting the "OTBN" block in OpenTitan. To reproduce:
```
git clone https://github.com/lowRISC/opentitan.git
cd opentitan
git checkout -b tst 77d8c98a78a668d0c7ea87ef0f84367f5cf8c656  # This is HEAD at the moment: probably other versions show it
fusesoc --cores-root=. run --target=lint --setup --build lowrisc:ip:otbn_top_sim
```
Alternatively, I've packed up the relevant code and attached it here: [verilator-bug.tar.gz](https://github.com/verilator/verilator/files/6502640/verilator-bug.tar.gz).

Download it and then run `test.sh` from the unpacked directory.

Before this commit, the lint job runs with no warnings. Afterwards:
```
> ./test.sh 
%Error: Internal Error: ../src/lowrisc_ip_otbn_top_sim_0.1/otbn_top_sim.sv:103:55: ../V3EmitC.cpp:2056: Wide Op w/ no temp, perhaps missing op in V3EmitC?
  103 |     assign dmem_rdata_nointeg[i*32 +: 32] = dmem_rdata[i*39 +: 32];
      |                                                       ^
                        ... See the manual and https://verilator.org for more assistance.
```

wsnyder, 2021-05-18:
@gezalore

gezalore, 2021-05-18:
Thanks @rswarbrick, I will investigate this shortly.

PR #2971 (2021-05-18): Don't merge bit select assignments in C code
Fixes #2970. The bug is caused by the optimzation enabled by 4f36e3e, merging the temporary added in V3Premit for wide ops into a concatenation when running `V3Const::constifyCpp` after V3Premit. This fix simply disables this optimisation when running the constant folder on lowered C++ code.

@wsnyder Intuitively I am under the impression that AstSel should not be present on the left hand side of an assignment by this stage (as in C++ you can only assign to whole words), should not these have been converted to AstWordSel?
closed 2021-05-18 by wsnyder