verilator/verilator #2972: Incorrect MULTIDRIVEN warning on unpacked array

mlefebvre1, 2021-05-19:
A MULTIDRIVEN warning is produced, when using 2 different clocks (clk0 and clk1) on 2 different processes which drives independant elements on an unpacked array (b). Here is the most simple example of this :

module t (
   input logic    clk0,
   input logic    clk1,
   output logic   b[2]
);
always_ff @(posedge clk0) begin
   b[0] <= 1;
end
always_ff @(posedge clk1) begin
   b[1] <= 1;
end
endmodule

Running **verilator --cc t.sv** will produce the following message :

%Warning-MULTIDRIVEN: test_tb.sv:4:19: Signal has multiple driving blocks with different clocking: 'b'
                      test_tb.sv:11:4: ... Location of first driving block
   11 |    b[1] <= 1;
      |    ^
                      test_tb.sv:7:4: ... Location of other driving block
    7 |    b[0] <= 1;
      |    ^
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=4.202
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)

I believe this is a bug as both instance of "b" should be considered independent and could be driven by different clocks.

mlefebvre1, 2021-05-19:
i've added a test for this issue : https://github.com/mlefebvre1/verilator/tree/issue-2972

wsnyder, 2021-05-19:
Thanks. This behavior is intentional as when writing logic that receives such a signal it is easy to have a clock domain crossing or timing bug (since the signals must not be treated the same based on bit position).  I've added clarification to the documentation.

closed 2021-05-19 by wsnyder