verilator/verilator #2973: BLKSEQ missed when connecting module port to array

Julian2n7000, 2021-05-19:
> Thanks for taking the time to report this.
>
> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
module DoesBlockingAssignA (
    input wire clk,
    output reg out
);

    always @(posedge clk)
        out = 1;

endmodule

module DoesBlockingAssignB (
    input wire clk,
    output reg out
);

    always @(posedge clk)
        out = 1;

endmodule

module TestBlockingAssign (
    input wire clk,

    output reg single,
    output reg array [1:0]
);

    DoesBlockingAssignA a( .clk(clk), .out(single) );

    DoesBlockingAssignB b( .clk(clk), .out(array[0]) );

endmodule
```

When this example is run under `verilator -Wall --lint-only`, `BLKSEQ` should be reported for both `DoesBlockingAssignA` and `DoesBlockingAssignB`. However `DoesBlockingAssignB` does not generate a `BLKSEQ` warning.

Specifically I am finding that if you perform a blocking assignment to a module port in a sequential block, and then connect that module port to part of an array, the `BLKSEQ` warning will be missed.

The behavior is still the same using vectors (i.e. `reg [1:0] foo`) rather than arrays (i.e. `reg foo [1:0]`): if the module port is connected to part of it, then `BLKSEQ` is missed.

> What 'verilator --version' are you using?  Did you try it with the git master version?

I tried the `verilator/verilator:latest` Docker container to get a more recent version. The behavior is still the same. It reports:
```
$ docker run -ti verilator/verilator:latest --version
Verilator 4.203 devel rev v4.202-42-g9699192d
```

> Would you be willing to try to fix Verilator yourself with assistance?

Yes, but I have not contributed before, so it would take some time to familiarize myself with the code.

wsnyder, 2021-05-20:
I suspect there's an attribute that gets lost in this case.  Thanks for offering to take a look.  If you run with --debug you can see the structure of the internal Ast tree, e.g. the VAR (variable) that is "single".  The code in V3Active.cpp then recurses this tree looking for the appropriate warning.

The other step once debugging would be to change the example you have into a test - basically just put it into test_regress as e.g. t_lint_blkseq_bad.v, and make a t_lint_blkseq_bad.pl file that looks like t_array_list_bad.pl.  Then run "test_regress/t/t_lint_blkseq_bad.pl --gold" and check that the file it makes (test_regress/t/t_lint_blkseq_bad.out) matches the output you expect.

Julian2n7000, 2021-05-23:
After trying the test under gdb, it seems the difference between the good and bad case with respect to the BLKSEQ check in V3Active.cpp is whether `isTemp()` is true in the outer if.

https://github.com/verilator/verilator/blob/7ea014dab53e9341ed2bda0a76d948d4029546ce/src/V3Active.cpp#L379-L382

Looking at the AST dumps, what I'm understanding is that after the 024_tristate pass, the port connection to the array reference gets replaced by a connection to a temporary signal (a MODULETEMP variable type), which is separately connected to the array through a new ASSIGNW node. Then after 026_inline, things are flattened and the module is inlined so the assignment refers to the temporary signal.

Julian2n7000, 2021-05-29:
@wsnyder, 

It looks like the `!varp->isTemp()` check was added in your commit 326cc8fd674aa2873d8107db57421f09d1574f64. You also added a test for calling a function inside a sequential block (in t_lint_blksync_bad). So the check avoids warning after inlining a function introduces blocking assignments into a sequential block.

Wondering what you would suggest as the fix. Can the check be made more specific (something like using `isFuncLocal` rather than `isTemp`) to avoid the warning only in the function case but not assigning to module ports, or are there other kinds of temporaries currently covered by it that should not generate warnings ?