verilator/verilator #2978: error @ Make test

sreersr, 2021-05-24:
Hi,
While installing verilator, I am getting 10 errors while running **make test**

%Error: export VERILATOR_ROOT=/home/sree/verilator/test_regress/..
%Error: setarch --addr-no-randomize gdb /home/sree/verilator/test_regress/../verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --debugi 0 --gdbbt --no-dump-tree -V" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed setarch --addr-no-randomize gdb /home/sree/verilator/test_regress/../verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --debugi 0 --gdbbt --no-dump-tree -V" -ex 'set width 0' -ex 'bt' -ex 'quit'
Use of uninitialized value $param{"logfile"} in concatenation (.) or string at ./driver.pl line 1639.
%Warning: vlt/t_a1_first_cc: Exec of perl failed: 

vlt/t_a1_first_cc: %Error: Exec of perl failed: 
vlt/t_a1_first_cc: FAILED: Exec of perl failed: 
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:01

======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of perl failed: 
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:01
make: *** [Makefile:254: smoke-test] Error 10

Please help me in this regard

wsnyder, 2021-05-24:
Did Verilator build successfully before this?  It looks like it's aborting immediately which is unusual but indicates something much more wrong than normal.  Is there something unusual about your system - what version of verilator, what distro, etc?

sreersr, 2021-05-24:
Hi,
Thanks for your email.
Version of verilator is 4.202 2021-04-24 rev v4.202-17-gaac4b709

Since I am failing in *make** test* step, I couldn't *build *it so far.
Please help me in this regards
Thanks in advance.

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

sreersr, 2021-05-24:
Hi Wilson,
this is the error after make test

#dist/t_a4_examples: %Error: Exec of make -C ../examples/cmake_hello_c
failed:
make -j && test_regress/t/t_a4_examples.pl  --dist
#vlt/t_flag_make_cmake_sc: %Error: Exec of cmake failed: make[2]: Entering
directory '/tmp/verilator/test_regress/obj_vlt/t_flag_make_cmake_sc'
make -j && test_regress/t/t_flag_make_cmake_sc.pl  --vlt
#vlt/t_savable_open_bad2: %Error: Exec of
obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error:
/tmp/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
make -j && test_regress/t/t_savable_open_bad2.pl  --vlt
#vltmt/t_flag_make_cmake_sc: %Error: Exec of cmake failed: make[2]:
Entering directory
'/tmp/verilator/test_regress/obj_vltmt/t_flag_make_cmake_sc'
make -j && test_regress/t/t_flag_make_cmake_sc.pl  --vltmt

*TESTS DONE, FAILED: Passed 2568  Failed 4  Failed-First 4  Skipped 2
 Unsup 30  Time 115:11Makefile:51: recipe for target 'test' failedmake[1]:
*** [test] Error 10make[1]: Leaving directory
'/tmp/verilator/test_regress'Makefile:258: recipe for target 'test_regress'
failedmake: *** [test_regress] Error 2*

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

wsnyder, 2021-05-25:
This is a pretty odd list of failures.  What OS distribution?   Is there anything special about your environment?  Did you modify anything in Verilator itself?

If nothing obvious above, you'll need to debug it.  For example to run a single test use e.g. "test_regress/t/t_flag_make_cmake_sc.pl  --vlt" and see if it's obvious what is going wrong.

wsnyder, 2021-06-06:
Didn't hear back - I'm closing assuming you found the problem - if you need additional assistance feel free to post more here.

closed 2021-06-06 by wsnyder