verilator/verilator #2980: Request binary release in GitHub Action

sequencer, 2021-05-25:
What would you like added/supported?
We wanna use verilator in our(Chisel) GitHub CI. However verilator didn't provide a binary release.

May we assist you in trying to fix this yourself?
If upstream think is possible, I may create a PR for verilator and publish it via GitHub Action.

wsnyder, 2021-05-25:
I was going to suggest to use docker, but... ;)

Can you describe please the process and where files would be put?  Also I would think your CI would want to use a submodule of verilator pegged to a last known good version, rather than a mainline build?

udif, 2021-05-25:
I just spent last week building some github actions for a few projects. I can help here if required.

@sequencer, if you already have a github actions CI running, why don't you build your own copy as part of it? You can easily cache it if you like, and you'll have more control over which version you pick. Building Verilator should take only a few shell commands and is trivial to add to an existing github action.

sequencer, 2021-05-26:
I wanna have a GitHub Action to install a specific verilator version with this in GitHub Action:
```yaml
- name: Install verilator 
  uses: xxx/setup-verilator 
  with: 
    version: 4.xxx
```
So we don't need to build and cache anymore.
except verilator binary, I think it also contains these things:
```
verilator /usr/
verilator /usr/bin/
verilator /usr/bin/verilator
verilator /usr/bin/verilator_bin
verilator /usr/bin/verilator_bin_dbg
verilator /usr/bin/verilator_coverage
verilator /usr/bin/verilator_coverage_bin_dbg
verilator /usr/bin/verilator_gantt
verilator /usr/bin/verilator_profcfunc
verilator /usr/share/pkgconfig/
verilator /usr/share/pkgconfig/verilator.pc
verilator /usr/share/verilator/
verilator /usr/share/verilator/bin/
verilator /usr/share/verilator/bin/verilator_includer
verilator /usr/share/verilator/include/
verilator /usr/share/verilator/include/gtkwave/
verilator /usr/share/verilator/include/gtkwave/fastlz.c
verilator /usr/share/verilator/include/gtkwave/fastlz.h
verilator /usr/share/verilator/include/gtkwave/fst_config.h
verilator /usr/share/verilator/include/gtkwave/fstapi.c
verilator /usr/share/verilator/include/gtkwave/fstapi.h
verilator /usr/share/verilator/include/gtkwave/lz4.c
verilator /usr/share/verilator/include/gtkwave/lz4.h
verilator /usr/share/verilator/include/gtkwave/wavealloca.h
verilator /usr/share/verilator/include/verilated.cpp
verilator /usr/share/verilator/include/verilated.h
verilator /usr/share/verilator/include/verilated.mk
verilator /usr/share/verilator/include/verilated.v
verilator /usr/share/verilator/include/verilated_config.h
verilator /usr/share/verilator/include/verilated_config.h.in
verilator /usr/share/verilator/include/verilated_cov.cpp
verilator /usr/share/verilator/include/verilated_cov.h
verilator /usr/share/verilator/include/verilated_cov_key.h
verilator /usr/share/verilator/include/verilated_dpi.cpp
verilator /usr/share/verilator/include/verilated_dpi.h
verilator /usr/share/verilator/include/verilated_fst_c.cpp
verilator /usr/share/verilator/include/verilated_fst_c.h
verilator /usr/share/verilator/include/verilated_heavy.h
verilator /usr/share/verilator/include/verilated_imp.h
verilator /usr/share/verilator/include/verilated_intrinsics.h
verilator /usr/share/verilator/include/verilated_save.cpp
verilator /usr/share/verilator/include/verilated_save.h
verilator /usr/share/verilator/include/verilated_sc.h
verilator /usr/share/verilator/include/verilated_sym_props.h
verilator /usr/share/verilator/include/verilated_syms.h
verilator /usr/share/verilator/include/verilated_threads.cpp
verilator /usr/share/verilator/include/verilated_threads.h
verilator /usr/share/verilator/include/verilated_trace.h
verilator /usr/share/verilator/include/verilated_trace_imp.cpp
verilator /usr/share/verilator/include/verilated_vcd_c.cpp
verilator /usr/share/verilator/include/verilated_vcd_c.h
verilator /usr/share/verilator/include/verilated_vcd_sc.cpp
verilator /usr/share/verilator/include/verilated_vcd_sc.h
verilator /usr/share/verilator/include/verilated_vpi.cpp
verilator /usr/share/verilator/include/verilated_vpi.h
verilator /usr/share/verilator/include/verilatedos.h
verilator /usr/share/verilator/include/vltstd/
verilator /usr/share/verilator/include/vltstd/svdpi.h
verilator /usr/share/verilator/include/vltstd/vpi_user.h
verilator /usr/share/verilator/verilator-config-version.cmake
verilator /usr/share/verilator/verilator-config.cmake
``` 
Since it is not a simple binary release with with a lot of headers, and cmake file, I think docker may be better thing to do this job, but I'm still seeking a possibility to directly use verilator via GitHub Action. 

> if you already have a github actions CI running, why don't you build your own copy as part of it?

I think we should get rid of maintaining another docker environment. If upstream may support this, it would be good to save other teams time to setup yet another verilator docker or build & caching.

imphil, 2021-05-26:
Building binary releases that actually work across a wide range of systems is surprisingly tricky due to runtime dependencies.

If you're looking for distribution packages for Verilator to use in CI environments have a look at
https://build.opensuse.org/project/show/home:phiwag:edatools. There you can find versioned packages of Verilator for a variety of distributions. We use those binaries in OpenTitan and Ibex CI, for example: https://github.com/lowRISC/ibex/blob/master/ci/install-build-deps.sh#L30

wallento, 2021-05-26:
I also have some preliminary work of a github action, where you can use arbitrary verilator versions. It is currently only functioning when using fusesoc (https://github.com/librecores/ci-fusesoc-action), the non fusesoc version goes through similar revamping at the moment. That will pretty much allow what you are planning.

The main "issue" we are facing is the real reason why its not really sufficient to run verilator in some container and use the artifacts: You want or need the same build environment for the verilator run, the build and the execution.

There was some discussion around that before: https://github.com/verilator/verilator/issues/2357

wsnyder, 2021-06-06:
I think we've finished discussing this (thanks all), presuming it can now be closed. Feel free to ask more or reopen.

closed 2021-06-06 by wsnyder