verilator/verilator #2982: Error in testing - while running "docker run -ti verilator/verilator-buildenv test"

sreersr, 2021-05-26:
Hi Thanks for your concern help, I have successfully build verilator now. But while running "docker run -ti verilator/verilator-buildenv test" again failed with error as in attachment. Please help me in this regard.
Also is it required to do/build "Verilator Build Docker Container", please let me know.

Thanks in advance.
[image]

imphil, 2021-05-26:
You're explicitly passing a compiler through CC and CXX which doesn't exist. Do you have a specific reason to use exactly this version of clang?

sreersr, 2021-05-26:
Hi Philip,
Thanks, what should I do now. As this is not familiar to me, I dont have
any specific reason to use this version of clang just following the
guidelines.

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

imphil, 2021-05-26:
Let's take a step back: what are you trying to achieve? Just use Verilator? Use Verilator in docker specifically? Which instructions did you follow? What operating system are you on? Etc.

wallento, 2021-05-26:
The buildenv container does not know clang-10. You are using it the right way. The main reason this buildenv container exists is to run and test verilator builds against different compilers. Is there any particular reason you want to build it with that container?

sreersr, 2021-05-26:
Hi,
I wanna use a verilator to convert RISC-V code in BSV to convert it into
verilog. I am new to this and just following the guidelines in this link
https://verilator.org/guide/latest/install.html. To be honest I don't know
is it required to use docker or not. I am running in Ubuntu. Please help
me, Philip. Thanks in advance.

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

sreersr, 2021-05-26:
Hi,
I wanna use a verilator to convert RISC-V code in BSV to convert it into
verilog. I am new to this and just following the guidelines in this link
https://verilator.org/guide/latest/install.html. To be honest I don't know
is it required to use or not. I am running in Ubuntu. Please help me,
Stefan. Thanks in advance.

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

wallento, 2021-05-26:
Then don't use the docker buildenv and instead only do the Git Quick Install: https://verilator.org/guide/latest/install.html#git-quick-install

sreersr, 2021-05-26:
Thanks Stefan, So only quick install is to be done and no need for other
installation procedures thanks a lot.

With Best Regards,
Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"Sometimes you win, sometimes you learn, but you always have to move
forward"*

wallento, 2021-05-26:
No, exactly, please close the issue if that works for you

wsnyder, 2021-06-06:
Closing under assumption all's now good, feel free to post more if not.

closed 2021-06-06 by wsnyder