verilator/verilator #2983: Can't get the value that would be assigned in the next GEN-FOR iteration.

yanx21, 2021-05-27:
Hi,
I‘m using the release version of verilator-4.202 and here is my testcase:
```verilog
// t.v
module t (
    input   logic   clk
);

logic [7:0]    mx;
logic [7:0]    A[3];
logic [7:0]    B[3];

always @ (posedge clk) begin
    mx <= mx + 1'b1;
    $write("\n -- %d\n", mx);
    $write("A:   %d    %d    %d\n", A[0], A[1], A[2]);
    $write("B:   %d    %d    %d\n", B[0], B[1], B[2]);

    if (mx == 'd5) begin
        $finish;
    end
end

genvar i;
generate
    for (i = 0; i < 3; i++) begin
        if (i == 2) begin
            always @ (*) begin
                A[i] = mx;
                B[i] = mx;
            end
        end
        else begin
            always @ (*) begin
                A[i] = mx;
                B[i] = A[i+1]; // Use the value that would be assigned in the next GEN-FOR iteration.
            end
        end
    end
endgenerate

endmodule

```
And the output is:
```txt
 --   0
A:     0      0      0
B:     0      0      0

 --   1
A:     1      1      1
B:     0      1      1

 --   2
A:     2      2      2
B:     1      2      2

 --   3
A:     3      3      3
B:     2      3      3

 --   4
A:     4      4      4
B:     3      4      4

 --   5
A:     5      5      5
B:     4      5      5
```
In my RTL design, `B[0]` should be equal to `A[1]`, but it’s not the case.

Post part of the output cpp file here:
```cpp
// ...
    vlTOPp->t__DOT__mx = __Vdly__t__DOT__mx;
    vlTOPp->t__DOT__A[2U] = vlTOPp->t__DOT__mx;
    vlTOPp->t__DOT__B[2U] = vlTOPp->t__DOT__mx;
    vlTOPp->t__DOT__A[0U] = vlTOPp->t__DOT__mx;
    vlTOPp->t__DOT__B[0U] = vlTOPp->t__DOT__A[1U]; // B[0] is assigned by A[1]
    vlTOPp->t__DOT__A[1U] = vlTOPp->t__DOT__mx; // A[1] is ready.
    vlTOPp->t__DOT__B[1U] = vlTOPp->t__DOT__A[2U];
// ...
```
As you can see, the `B[0]` is assigned by `A[1]`  before `A[1]` is ready. This must be wrong.

I guess this is a bug of Verilator, what's your opinion.

yanx21, 2021-06-03:
@wsnyder Any comment on this?

udif, 2021-06-03:
Just to make things interesting, I made minor modifications to your test:
* added initial value to mx
* replaced `i++` with `i=i+1` for iverilog
* added an internal clock signal `clk` instead of a clock port

```
module t (
);

reg clk;
initial clk = 1'b0;

  always
    clk = #5 ~clk;
  
logic [7:0]    mx;
logic [7:0]    A[3];
logic [7:0]    B[3];

  initial 
    mx = 0;
  
always @ (posedge clk) begin
    mx <= mx + 1'b1;
    $write("\n -- %d\n", mx);
    $write("A:   %d    %d    %d\n", A[0], A[1], A[2]);
    $write("B:   %d    %d    %d\n", B[0], B[1], B[2]);

    if (mx == 'd5) begin
        $finish;
    end
end

genvar i;
generate
  for (i = 0; i < 3; i = i + 1) begin
        if (i == 2) begin
            always @ (*) begin
                A[i] = mx;
                B[i] = mx;
            end
        end
        else begin
            always @ (*) begin
                A[i] = mx;
                B[i] = A[i+1]; // Use the value that would be assigned in the next GEN-FOR iteration.
            end
        end
    end
endgenerate

endmodule
```

I ran it on edaplayground and the output, for Riviera Pro is:
```
# KERNEL:  --   0
# KERNEL: A:     0      0      0
# KERNEL: B:     0      0      0
# KERNEL: 
# KERNEL:  --   1
# KERNEL: A:     1      1      1
# KERNEL: B:     1      1      1
# KERNEL: 
# KERNEL:  --   2
# KERNEL: A:     2      2      2
# KERNEL: B:     2      2      2
# KERNEL: 
# KERNEL:  --   3
# KERNEL: A:     3      3      3
# KERNEL: B:     3      3      3
# KERNEL: 
# KERNEL:  --   4
# KERNEL: A:     4      4      4
# KERNEL: B:     4      4      4
# KERNEL: 
# KERNEL:  --   5
# KERNEL: A:     5      5      5
# KERNEL: B:     5      5      5
```

icarus 0.10.0 is similar (without the `# KERNEL:` prefix on the output)

yanx21, 2021-06-03:
@udif Yes, many other simulators give identical result, which I think would be the correct behaviors. And this issue should be a bug of verilator. Looking forward to your reply @wsnyder

wsnyder, 2021-06-03:
Think this should have gotten an UNOPTFLAT warning and the appropriate code have been inserted to reevaluate this.

Perhaps you'd be willing to dig in and see why this isn't happening?  Otherwise I will take a look later.

yanx21, 2021-06-03:
> Think this should have gotten an UNOPTFLAT warning and the appropriate code have been inserted to reevaluate this.
> 
> Perhaps you'd be willing to dig in and see why this isn't happening? Otherwise I will take a look later.

There is no any warning or error.

What's more, I removed the FOR statement and built simpler cases that may be more helpful.
Post 3 cases here. Transform and running results are on the comment.
``` systemverilog
module t (
    input   logic   clk
);

logic [7:0]    mx;
logic [7:0]    A[3];
logic [7:0]    B[3];

always @ (posedge clk) begin
    mx <= mx + 1'b1;
    $write("\n -- %d\n", mx);
    $write("A:   %d    %d    %d\n", A[0], A[1], A[2]);
    $write("B:   %d    %d    %d\n", B[0], B[1], B[2]);

    if (mx == 'd5) begin
        $finish;
    end
end

always_comb begin
// #1: No warning
//       Bad runing result
    A[0] = mx;
    B[0] = A[1];

    A[1] = mx;
    B[1] = A[2];

    A[2] = mx;
    B[2] = mx;

// #2: No warning
//       Good running result
//     A[2] = mx;
//     B[2] = mx;
//
//     A[1] = mx;
//     B[1] = A[2];
//
//     A[0] = mx;
//     B[0] = B[1];

// #3: %Warning-ALWCOMBORDER and %Warning-UNOPTFLAT
//       Good running result if I disable the warnings.
//     A[0] = mx;
//     A[1] = mx;
//     A[2] = mx;
// 
//     B[0] = B[1];
//     B[1] = A[2];
//     B[2] = mx;
end

endmodule
```

gezalore, 2021-06-04:
This issue looksrelated to #2576

yanx21, 2021-08-09:
Any other thoughts on this issue?

wsnyder, 2025-09-08:
I believe the output currently matches what master produces, so fixed somewhere along the line, if there are still related problems please open a new issue, thanks.

closed 2025-09-08 by wsnyder