verilator/verilator #2984: The out-of-bounds array access behavior is inconsistent with that of the VCS.

Superchao-ai, 2021-05-27:
Hi,
I‘m using the release version of verilator-4.202 and here is my example:
`
module top(clk);
	input clk;

	logic [15:0] index;
	logic [15:0] data;
	logic [15:0] mem[7:0];

	always @(posedge clk)
		begin
			data  <= data + 1'b1;
			index <= index + 1'b1;
			mem[index] <= data;  // index will be out of range
		end

endmodule
`

I notice that the verilator output cpp is as follows:
`
VL_INLINE_OPT void Vtop::_sequent__TOP__3(Vtop__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop::_sequent__TOP__3\n"); );
    Vtop* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Variables
    // Begin mtask footprint all: 
    CData/*2:0*/ __Vdlyvdim0__top__DOT__mem__v0;
    SData/*15:0*/ __Vdly__top__DOT__data;
    SData/*15:0*/ __Vdly__top__DOT__index;
    SData/*15:0*/ __Vdlyvval__top__DOT__mem__v0;
    // Body
    __Vdly__top__DOT__index = vlTOPp->top__DOT__index;
    __Vdly__top__DOT__data = vlTOPp->top__DOT__data;
    __Vdly__top__DOT__data = (0xffffU & ((IData)(1U) 
                                         + (IData)(vlTOPp->top__DOT__data)));
    __Vdly__top__DOT__index = (0xffffU & ((IData)(1U) 
                                          + (IData)(vlTOPp->top__DOT__index)));
    __Vdlyvval__top__DOT__mem__v0 = vlTOPp->top__DOT__data;
    __Vdlyvdim0__top__DOT__mem__v0 = (7U & (IData)(vlTOPp->top__DOT__index));
    vlTOPp->top__DOT__data = __Vdly__top__DOT__data;
    vlTOPp->top__DOT__index = __Vdly__top__DOT__index;
    vlTOPp->top__DOT__mem[__Vdlyvdim0__top__DOT__mem__v0] 
        = __Vdlyvval__top__DOT__mem__v0;
}
`
`__Vdlyvdim0__top__DOT__mem__v0 = (7U & (IData)(vlTOPp->top__DOT__index));`
this code shows that only the lower 3-bit used as the index.
If the index exceeds the depth of the array, will incorrectly assigned to the array.
In VCS simulation, indexes that exceed the array depth are ignored and no value is assigned.

Can you solve this problem by keeping the simulation behavior of the verilator consistent with that of the VCS?

wsnyder, 2021-05-29:
The behavior you want is required by IEEE, so this is a bug.

Verilator does simplify the range checks for performance, I need to look into this case to see if this case can be properly handled without significant performance loss.  I suspect not.

PeterMonsson, 2021-05-29:
I remember this as a simulation vs. synthesis mismatch issue. Consider adding a warning for this case.

gezalore, 2021-05-29:
FWIW, compilers seem to turn `a >= C  ? C : a` into a branch free sequence with `cmov` (on x86), or `csel` (on arm):
https://godbolt.org/z/TW8GfcKvx
so it might not be too bad if you just make all arrays 1 larger than declared, and dump out of range values in the spare entry:
```
vlTOPp->top__DOT__mem[__Vdlyvdim0__top__DOT__mem__v0 >= 8 ? 8 : __Vdlyvdim0__top__DOT__mem__v0] = ...
```

margej, 2021-06-18:
I suspect many synthesis tools behave in the same way here. 
Rather than forfeit performance for strict IEEE compliance, which may not match a synthesised design, as @PeterMonsson said, I think a lint warning may be a better approach?

wsnyder, 2021-09-12:
I suspect a new warning, but no time has been put into this recently.  If you or someone would like to look into it, that would be appreciated.