verilator/verilator #2988: Emit global functions instead of methods.

gezalore, 2021-05-28:
I have been working on #2958, and have an initial implementation of the hash based function splitting in place. For a sneak peek you can look at V3Order here: https://github.com/gezalore/verilator/commit/41f1aa3ff6811f61bc698fbb7429e1263d42e4f4. This works really well in the sense that adding or removing statements causes only a handful of output functions to change as expected. 

Also as expected, sadly this is still useless due to the issues described in #2140. In this issue I would like to describe my complete proposal for solving the first problem descibed in #2140, namely the problem that methods are declared in the model header, which means a recompile of everything is still needed when a function is added/removed (or changes content due to hash based naming). The idea is basically the same as was proposed in #2140, but I am describing it in full separately here for approval.

Here are my proposed changes to the emitted code and to internals:

1. Verilator should only emit global functions, which take at lease one, or both of the following arguments:
  a. An argument `ModuleClass* self`, (name subject to change if you would like to) being the explicitly passed replacement for `this`, if the function uses relative references (as defined in V3Descope).
  b. `Syms* vlSymsp`, if the function uses absolute references(as defined in V3Descope).
Additional arguments are of course fine where needed, I'm mostly talking about the main logic implementation functions here which only need these two. I however propose that in order to keep Verilator internals simple, we use this pattern for everything, even translated SV classes. It seems simpler to me to use only one representation, rather than having both global functions and methods intermixed, plus global functions allow us to do what I describe in point 3 below.

2. The only true methods generated in the output should be ones that the user is actually expected to call on the model (e.g.: main entry points and verilator public stuff), and ideally these would be thin wrapers generated in V3Emit that simply dispatch to the equivalent global function, passing `this` for `self`. These should be defined in the class header so they are inlined. I realize this might add a bit more dependency on the header, but I think I can fix this later when resolving the dependencies on the state variables in the header, and changing the public interface is hopefully rare in the common edit-compile-test loop.

3. The global functions emitted in point 1. should be declared lazily, only where they are actually called. This can be done by scanning emitted code for calls, and emitting a declaration before the function guarded by a preprocessor macro, e.g. when `bar` calls `foo`, we emit:
```
#ifndef VL_DECLARED__FOO
#defind VL_DECLARED__FOO
void foo(Module* self, Syms* vlSymsp);
#endif
void bar(Module* self, Syms* vlSymsp) {
  ...
  foo(self, vlSymsp);
  ...
}
```
The guard ifdef is necessary to keep support for compiling via the concatenated `*__All.cpp` file.

I actually have a basic proof of concept patch which implements a lot of point 1  (albeit still keeping methods around internally) and all of 3, and it works well so far. A lot of the 50-odd test failures remaining are due to `$c` int the tests, which need to be chagned to reference variables via `self`.

At this point I would like to ensure you are happy @wsnyder to proceed with this scheme before I put in the remaining effort to make it a proper patch as the remaining work is a bit involved.

wsnyder, 2021-05-29:
In principal this seems a smart idea that should work.

While generally making static functions seems OK, note that Verilog classes must still have member functions when they are virtual.  We could just keep those in the C++ class declaration rather than optimize them which seems fine - but this means you can't just simplify the code so everything is static unfortunately.

I believe it's perfectly legal to forward-declare the same function twice, so you should not need any ifdefs.

gezalore, 2021-05-29:
Good point about the virtuals, I forgot about that, thank you.

> I believe it's perfectly legal to forward-declare the same function twice, so you should not need any ifdefs.

You are right, but actually the declaration can appear after the definition as well, but that also seems ok. I was worring about some compiler warnings causing issues eg with `-Wextra -Werror`, but neither gcc nor clang issues any warnings that I could trigger for either multiple forward declarations, or repeat declarations after the definition, so I will proceed without guards initially, we can always add them back if we find we need to.

wsnyder, 2021-06-13:
Thanks for your large work on this, with #3006 merged believe this is done, yea!

closed 2021-06-13 by wsnyder