verilator/verilator #2989: MULTIDRIVEN warning, where TB and RTL write the same object

agrobman, 2021-05-28:
I have TB code,  slamming DUT registers for some simulation shortcuts, but verilator issues MULTIDRIVEN warning for this TB code. ( It also didn't like that the flop state variable had <= vs TB used '=' assignment. )

I'm slamming some library flop cell, I can't change.

Is there more intelligent way to disable it than global -Wno-MULTIDRIVEN ?

I've tried /* verilator lint_off MULTIDRIVEN */ around my TB code with no success ..

wsnyder, 2021-05-29:
You added the lint_off around the exact line number that Verilator issues the warning for?  This should work, as should -Wno-MULTIDRIVEN.

agrobman, 2021-05-29:
I placed the comments around TB code only, (I can't do this in design), but it didn't work

agrobman, 2021-05-29:
strange the following code compiles with no warnings:

```
module test ( input clk, dd);
logic data, din, dout;

task drive;
     dff.dout = 1;
endtask

rvdff dff(dd, clk, 1, dout);

endmodule

module rvdff #( parameter WIDTH=1 )
   ( 
     input logic [WIDTH-1:0] din,
     input logic           clk,
     input logic                   rst_l,
     output logic [WIDTH-1:0] dout
     );

   always_ff @(posedge clk or negedge rst_l) begin
      if (rst_l == 0)
        dout[WIDTH-1:0] <= 0;
      else
        dout[WIDTH-1:0] <= din[WIDTH-1:0];
   end
endmodule 
```

Although, I would think it should because of SV rules about always_ff

But  bigger design it does:

lines 428 - 467:
```
`define XREG(i) `DEC.arf.gpr[i].gprff.genblock.dff.dout

/* verilator lint_off MULTIDRIVEN */

function void slam_gpr(bit[4:0] rn, bit[`RV_XLEN-1:0] data);
    case(rn)
    1: `XREG(1) <= data;
    2: `XREG(2) <= data;
    3: `XREG(3) <= data;
    4: `XREG(4) <= data;
    5: `XREG(5) <= data;
    6: `XREG(6) <= data;
    7: `XREG(7) <= data;
    8: `XREG(8) <= data;
    9: `XREG(9) <= data;
    10: `XREG(10) <= data;
    11: `XREG(11) <= data;
    12: `XREG(12) <= data;
    13: `XREG(13) <= data;
    14: `XREG(14) <= data;
    15: `XREG(15) <= data;
    16: `XREG(16) <= data;
    17: `XREG(17) <= data;
    18: `XREG(18) <= data;
    19: `XREG(19) <= data;
    20: `XREG(20) <= data;
    21: `XREG(21) <= data;
    22: `XREG(22) <= data;
    23: `XREG(23) <= data;
    24: `XREG(24) <= data;
    25: `XREG(25) <= data;
    26: `XREG(26) <= data;
    27: `XREG(27) <= data;
    28: `XREG(28) <= data;
    29: `XREG(29) <= data;
    30: `XREG(30) <= data;
    31: `XREG(31) <= data;
    endcase
endfunction
/* verilator lint_on MULTIDRIVEN */
```
last warning looks:

```
%Warning-MULTIDRIVEN: design/lib/beh_lib.sv:248:31: Signal has multiple driving blocks with different clocking: 'tb_top.rvtop.swerv.dec.arf.__Vcellout__gpr[31].gprff__dout'
                      design/lib/beh_lib.sv:36:2: ... Location of first driving block
   36 |  dout[WIDTH-1:0] <= 0;
      |  ^~~~
                      /verif/vtb/tb_top.sv:464:56: ... Location of other driving block
  464 |     31: rvtop.swerv.dec.arf.gpr[31].gprff.genblock.dff.dout <= data;
      |                                                        ^~~~

```

agrobman, 2021-06-08:
Hi, any updates on this issue?  when /how does verilator detect MULTIDRIVENs? why does it warns "big" design and doesn't in the small?

wsnyder, 2022-10-15:
Looks like this got stuck.... what is the precise question or change suggested here?  If you are getting MULTIDRIVEN where unexpected I'm not sure what code sample shows the problem - can you reattach?

As to the last question, small designs might not show MULTIDRIVEN because in some cases Verilator may optimize away the conflict.  MULTIDRIVEN is checked relatively late, versus before optimization.

wsnyder, 2023-01-26:
Closed pending feedback.

closed 2023-01-26 by wsnyder