verilator/verilator #2993: Supports bit operation combination

flex-liu, 2021-05-31:
the verilog like this:
always @(*) begin
   for(int j = 0; j < 32; j= j +1) begin
      value[j] = in1_ff[j] & in2_ff[j];
  end
end
We want the generated C++ code to look like this:
 value = in1_ff & in2_ff
Combine the same operations as much as possible and cannot be directly translated into bit operations.

wsnyder, 2021-05-31:
Verilator will only optimize binary operations that look like this presently. I agree that doing better would be beneficial.

If you or someone would like to work on this we can discuss the algorithms etc required. This will be a fairly complicated undertaking.

ObiWanRohan, 2022-09-30:
@wsnyder Is there any place we can take ideas from? I think we could look into compilers like gcc or LLVM for algorithm and implementation ideas.

gezalore, 2022-09-30:
The DFG representation on the develop-v5 branch should be able to represent this (currently it does not convert `always_comb` though, so that would need to be extended. We could then write a pass to combine bitwise opts over adjacent selects.

ObiWanRohan, 2022-09-30:
What's `always_comb`?

For an initial implementation for the pass, I think combining these operations if the complete array is assigned could work. After that, it could be modified to work for consecutive subparts.

gezalore, 2022-09-30:
> What's always_comb?

Similar to `always @(*)` which are treated largely the same internally.