verilator/verilator #2996: asterisk not supported in command files

jotego, 2021-06-01:
Although you can do something like `verilator rtl/*.v` in the command line, this will cause an error when moved to a command file: `verilator -f foo.f`, where the file foo.f is
```
rtl/*.v
```
This results in the error:
```
%Error: Unterminated /* comment inside -f file.
```
I know that the asterisk expansion is done by the shell, but the behaviour is expected in command files too. This works in _ncverilog_, for example.

Could you consider adding this feature?

Thank you

wsnyder, 2021-06-01:
Thanks for your report, I can understand why this was confusing.

Note the -f support is not as "standard" as the name might suggest in areas like these. There's several issues were this to be fixed:
1. * expansion is done by the shell, not by Verilator, so even if the * was recognized it would not find your files.
2. Verilator wouldn't be able to differentiate it from a /*comment*/ as the error indicates.  Comments are very common.
3. Its generally bad practice to put * on any command file/command line, as it's too easy for a user to add a temporary file like e.g. "debugging.v" then get screwed by the wildcard.  Much better to use an e.g. -y and only specify the top filename.

My strong leaning is this will not be fixed, and marking it so, and just updating the documentation.

jotego, 2021-06-01:
Thank you. If it is documented, I can try to avoid too.

closed 2021-06-01 by jotego