verilator/verilator #2998: Unexpected error 255 when compiling RV core

aignacio, 2021-06-01:
Hi,
when trying to compile CV32E40P RISC-V core I'm seeing an unexpected error 255.
```bash
- V3Ast.cpp:1138:     Dumping output_verilator/Vsimple_tile_002_cells.tree
%Error: Internal Error: rtl/cv32e40p/rtl/cv32e40p_id_stage.sv:31:22: ../V3Broken.cpp:253: Broken link in node (or something without maybePointedTo): !m_packagep || !m_packagep->brokeExists()
-node: PACKAGEIMPORT 0x7f9a09d825c0 <e95652> {x31aw}  * -> PACKAGE 0x7f9a19eb2270 <e314138> {e26aj}  cv32e40p_pkg  L7 [LIB] [NONE]
   31 |   import cv32e40p_pkg::*;
      |                      ^~
                        ... See the manual and https://verilator.org for more assistance.
- V3Ast.cpp:1138:     Dumping output_verilator/Vsimple_tile_990_final.tree
- V3Broken.cpp:318:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:226:statsReport:
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
make: *** [output_verilator/Vsimple_tile.mk] Error 255
```
I'm adding the includes first and the packages but even though I'm seeing the error. Can you point me what's wrong with my setup?

**[Repository](https://github.com/aignacio/mpsoc_example)**
**Verilator version:** Verilator 4.106 2020-12-02 rev v4.106

wsnyder, 2021-06-01:
There's something it doesn't like about that import (probably that's obvious ;).  Can you please try to make a small standalone test case that shows this error?

aignacio, 2021-06-03:
Sorry, I'm not sure how to make a standalone test once it's the complete design who fails when compiling it....is there any information I can extract from the error by enabling some flags?

wsnyder, 2021-06-03:
Generally the easiest thing to do is start with the code that gives the error, and start deleting module instantiations, statements, etc making sure the error remains.  If the error disappears back up to undo your last edit.

wsnyder, 2021-08-24:
Closing as need more info, if you can provide a small test please feel free to reopen.

closed 2021-08-24 by wsnyder