verilator/verilator #2999: Non-GNU implementation of the ar command does not read command-line options from file

ahouska, 2021-06-02:
When generating the simulator, I'm getting this error:

```
echo "" > Vnaivecpu__ALL.verilator_deplist.tmp
Archive ar -rcs Vnaivecpu__ALL.a Vnaivecpu__ALL.o
ar: warning: can't open file: @Vnaivecpu__ALL.a.verilator_deplist.tmp: No such file or directory
```

The problem seems to be that the `ar` command on FreeBSD does not support the '@' prefix indicating that the file contains command-line options.

Perhaps `ar` should be called (from verilated.mk) multiple times if there are too many files to be added?

wsnyder, 2021-06-04:
The reason for the filename was otherwise the command line is too long. Not sure how to call ar multiple times to get around that.

Perhaps we need a wrapper around ar just to appease freebsd?   @ahouska would you be willing to look into doing that or another fix for freebsd?

mmicko, 2021-06-04:
can confirm this is also problem for macOS since they are using BSD implementation of `ar` as well.

ahouska, 2021-06-18:
@wsnyder I realize that the GNU implementation of `ar` can be installed on FreeBSD too (with the `binutils` package). So I think the solution is to adjust the `configure` script so it checks the `ar` implementation and allows user to specify the path to the GNU implementation. I can try but need to learn a bit about `configure`.

wsnyder, 2021-06-19:
Good suggestion, "./configure AR=whatever" will now override ar.o

yurivict, 2021-09-30:
I've added the ```AR=/usr/local/bin/ar``` to the ```cad/verilator``` port.

This issue can probably be closed.

closed 2021-09-30 by wsnyder

quantrpeter, 2021-10-31:
in mac, same problem. @yurivict your way not working in mac

yurivict, 2021-10-31:
I don't use mac. You need to install the GNU ar command. On FreeBSD it is installed by the package as ```/usr/local/bin/ar```