verilator/verilator #3001: Wrong version reported with `--version`?

kaddkaka, 2021-06-02:
I just downloaded verilator 4.202, unzipped it and executed `verilator-4.202/bin/verilator --version`, but my output is "Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd" why could that be?

My old binary was that version but I even renamed it: `/usr/bin/verilator` ->  `/usr/bin/verilator_old`

wsnyder, 2021-06-02:
You must still have the old version somewhere, see what "which verilator" shows, and/or "printenv" and look for VERILATOR settings.

closed 2021-06-02 by wsnyder

wsnyder, 2021-06-02:
Then that file is wrong, maybe you mis-built verilator - start from scratch again?  There's no way verilator itself has a problem related with this, it must be something related to your building or installing.

wallento, 2021-06-02:
> I just downloaded verilator 4.202

Do you have the link that you downloaded?

kaddkaka, 2021-06-02:
I did not do any build or install steps, I downloaded from the release page and unzipped this one: https://github.com/verilator/verilator/archive/refs/tags/v4.202.zip

wallento, 2021-06-02:
That's not a build, but "only" the source code. This is a wrapper around the binary. What is your output for `which verilator_bin`?

kaddkaka, 2021-06-02:
oh ok, that explains it, then I should build it, thanks.