verilator/verilator #3002: @(posedge clk)

HamzaShabbir517, 2021-06-03:
I have design a testbench in system verilog and i am simulating it on verilator. I have a logic of while loop which says

while(condition false) begin
   @(posedge clk);
   x = y;
end

This syntax is seem to be correct according to system verilog language but why it is giving me syntax error on verilator

HamzaShabbir517, 2021-06-03:
I have design a testbench in verilog and i am simulating it on verilator. I have a logic of while loop which says

while(condition false) begin
@(posedge clk);
x = y;
end

I am getting unexpected syntax error in verilator

wsnyder, 2021-06-03:
Verilator is a mostly cycle-based simulator and does not support event expressions like this.  You need to write it in a synthesizable form e.g. "always @ (posedge clk) ...".

closed 2021-06-03 by wsnyder

HamzaShabbir517, 2021-06-03:
I have try that but it was giving error at always block saying unexpected as it was inside the while loop. Any other solution

udif, 2021-06-03:
Your problem is that you are trying to write your testbench is SystemVerilog.
The current version of Verilator works best when you only run your synthesizable code through it, and write your testbench in C++.
Take a look for example at [https://github.com/verilator/verilator/blob/master/examples/make_tracing_c/sim_main.cpp](https://github.com/verilator/verilator/blob/master/examples/make_tracing_c/sim_main.cpp)

HamzaShabbir517, 2021-06-03:
No it is written in the verilog

udif, 2021-06-03:
Yes, I see it is written in Verilog.
The code you included is not synthesizable, or at least not written in synthesizable style. I don't know if it's a part of your testbench or your DUT.

If it's it's part of your DUT you need to fix your coding style to use synthesizable constructs.
If it's part of your tests, you'll have to rewrite them in C++.

HamzaShabbir517, 2021-06-03:
ok thanks for the response

HamzaShabbir517, 2021-06-04:
@udif 
Does the same syntax is also not allowed to used in task?
As i have gone through document they say in verilog we can used always block but i am using it and simulating it on verilog it is giving me same syntax error. Why is that so? Is there any way that i can use a flag to stop this warning

wsnyder, 2021-06-04:
It's an error not a warning, so you cannot stop it.  Verilator supports mostly only synthesizable always blocks. This sort of event control is not synthesizable.  You must use instead a e.g. finite state machine as you would write if you were to synthesize it.

HamzaShabbir517, 2021-06-04:
but i only want to use it for the testing purpose so what should i do. As i have a task which have to send 1 bit per clock. As verilator is not supporting that so i am stuck in that. Any possible solution

wsnyder, 2021-06-04:
I'm not sure what is not clear.  As said above Verilator doesn't support what you want, and you need to recode the Verilog if you want to use Verilator, or use a different tool than Verilator.

HamzaShabbir517, 2021-06-04:
ok