verilator/verilator #3005: OoM Verilating code with accidental large replication operation

margej, 2021-06-03:
Something similar to the following code caused an out-of-memory error when evaluated as part of a larger design. In this example allocation peaked at a little over 12GB on my CentOS7 system:

```
module t #( parameter NV = 2500) (
    input          a,
    input          w1,
    input  [127:0] w2,
    output [ 31:0] o
);

wire [NV-1:0] d = a ? NV'(0) : {NV{w2}};
assign        o = d[31:0];

endmodule
```
Version 4.102 was used with the following command line:
```
verilator --cc t.v
```

The issue was due to a coding error where w1 should have been used instead of w2, and the WIDTH warning emitted makes this clear, but I was wondering if there was an underlying issue under the hood causing the allocation explosion?

margej, 2021-06-03:
The issue appears to occur in V3Expand

wsnyder, 2021-06-04:
It's definitely over-aggressive. I'll add a limit to prevent this insanity.

closed 2021-06-06 by wsnyder