verilator/verilator #3010: enumeration of type struct packed, using structure assignment patterns, fails to compile

jeras, 2021-06-05:
I have created the tetcase `t_enum_struct`, it can be taken from my branch:
https://github.com/jeras/verilator/tree/enum_struct
https://github.com/jeras/verilator/blob/enum_struct/test_regress/t/t_enum_struct.v

The test works properly if a simple concatenation is used instead of an assignment pattern.

I checked the example with a few professional simulators and they seem to indicate, using a packed structure as the base type of an enumeration is illegal. I do not understand the formal definition of the enumeration type (standard 6.19 Enumerations) well enough to tell if this is legal or not.

Now I am not sure if it would make sense to add functionality or report an error "illegal enumeration base type".

In the given example, the use of a structure for the enumeration base type makes sense. But it is true, I wrote this code with the intention to heavily use SystemVerilog RTL features, beyond anything I saw in other SystemVerilog examples.

wsnyder, 2021-06-05:
An enum_base_type must be a integer atom or vector type, or typedef of this, so yes, this is illegal.

Perhaps you would consider making a pull request to check for this?  Basically in V3Width look at the enum's dtypeSkipRef() and complain if it isn't a BasicDType.  Then have the test (which you can simplify) check that the error properly comes out.

jeras, 2021-06-05:
Since:
* this is something only I use,
* it is not of high priority,
* from your description it sounds trivial.
So yes, I will try.

closed 2024-11-05 by wsnyder