verilator/verilator #3012: read internal register in gate-level

yangyang0417, 2021-06-09:
Hello!
I first used Synopsys Design Compiler to synthesize a RTL design to gate level, and then I want to access a 4-bit register signal, e.g. **temp[3:0]**, through testbench(already marked that signal as /*verilator public*/). However in the gate level, **temp[3:0]** is converted to 4 flip-flops respectively(**temp_reg[0], temp_reg[1], temp_reg[2], temp_reg[3]**). Assuming top module is **v**, submodule is **a** and **a** contains register **temp**, why I failed to read the value of temp by top->v->a->v ? Should I access each **temp_reg** respectively and then add them together?

wsnyder, 2021-06-09:
I suspect your gate netlist has signals named e.g. "\temp_reg[0] " that is the 0 is part of the signal name.  You'd then need to get at the public signal as "temp_reg__BRA__0__KET__".  Best thing is to look at the generated C code to see what the precise name is.

Alternatively, in Verilog make a e.g. DPI export function that reads the appropriate signal using Verilog, then you don't need to worry about the verilator public naming conventions etc, and it will also work on other simulators.

closed 2021-06-09 by wsnyder

yangyang0417, 2021-06-10:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Thanks! A follow up question, by passing --bbox-unsup, does verilator just ignore table in verilog or it can generate correct output according to the table, because I notice that handbook says, "passing --bbox-unsup will likely cause incorrect simulation".Thanks again for your help and patience!Best,Mingyang
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                mingyangduan0417
                            
                    
                        
                            
                                    ***@***.***
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

wsnyder, 2021-06-10:
Anything blacked box like tables will not get correct results.

yangyang0417, 2021-06-10:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Thanks so much and wish you have a nice day!Best,
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                mingyangduan0417
                            
                    
                        
                            
                                    ***@***.***
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制