verilator/verilator #3014: Support for --top-module overriding toplevel modules within a file

dpetrisko, 2021-06-10:
Thanks for taking the time to report this.

What would you like added/supported?
Currently, modules which are not a toplevel within their file cannot be set as toplevel through --top-module. The use-case here is to employ something like 'abstract modules' in SystemVerilog. See the comment below, but the takeaway is that modules which are instantiated are not considered toplevels by systemverilog parsers, and therefore do not elaborate. This is desired behavior for modules which have default parameters which elaborate to large or difficult cases. In VCS, it is possible to still designate the module as a toplevel, even though it is not by default inferred as a toplevel due to the abstract parameter.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
```
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_mux.v
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_misc/bsg_defines.v

// place this macro at the end of a verilog module file if that module has invalid parameters
// that must be specified by the user. this will prevent that module from becoming a top-level
// module per the discussion here: https://github.com/SymbiFlow/sv-tests/issues/1160 and the
// SystemVerilog Standard

//    "Top-level modules are modules that are included in the SystemVerilog
//    source text, but do not appear in any module instantiation statement, as
//    described in 23.3.2. This applies even if the module instantiation appears
//    in a generate block that is not itself instantiated (see 27.3). A design
//    shall contain at least one top-level module. A top-level module is
//    implicitly instantiated once, and its instance name is the same as the
//    module name. Such an instance is called a top-level instance."
//  

`define BSG_ABSTRACT_MODULE(fn) module fn``__abstract(); if (0) fn not_used(); endmodule

// macro for defining invalid parameter; with the abstract module declaration
// it should be sufficient to omit the "inv" but we include this for tool portability
// if later we find that all tools are compatible, we can remove the use of this from BaseJump STL

//`define BSG_INV_PARAM(param) param = "inv" 
`define BSG_INV_PARAM(param) param

module bsg_mux #(`BSG_INV_PARAM(width_p)
                 , els_p=1
                 , harden_p = 0
                 , balanced_p = 0
                 , lg_els_lp=`BSG_SAFE_CLOG2(els_p)
                 )
   (
    input [els_p-1:0][width_p-1:0] data_i
    ,input [lg_els_lp-1:0] sel_i
    ,output [width_p-1:0] data_o
    );
   
   if (els_p == 1)
     begin
      assign data_o = data_i;
      wire unused = sel_i;
     end
   else
     assign data_o = data_i[sel_i];

   // synopsys translate_off
   initial
     assert(balanced_p == 0)
       else $error("%m warning: synthesizable implementation of bsg_mux does not support balanced_p");
   // synopsys translate_on

endmodule

`BSG_ABSTRACT_MODULE(bsg_mux)
```

Result:
```
$ verilator --version
Verilator 4.111 devel rev v4.110-8-g4fa41c5d
$ verilator --cc --build bsg_defines.v bsg_mux.v --top-module bsg_mux
%Warning-PINMISSING: bsg_mux.v:30:44: Cell has missing pin: 'data_i'
   30 | module bsg_mux__abstract(); if (0) bsg_mux not_used(); endmodule
      |                                            ^~~~~~~~
                     ... Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: bsg_mux.v:30:44: Cell has missing pin: 'sel_i'
   30 | module bsg_mux__abstract(); if (0) bsg_mux not_used(); endmodule
      |                                            ^~~~~~~~
%Warning-PINMISSING: bsg_mux.v:30:44: Cell has missing pin: 'data_o'
   30 | module bsg_mux__abstract(); if (0) bsg_mux not_used(); endmodule
      |                                            ^~~~~~~~
%Error: Specified --top-module 'bsg_mux' isn't at the top level, it's under another instance 'bsg_mux__abstract'
%Error: Exiting due to 1 error(s), 3 warning(s)
```

Is this a verilator-specific limitation or are we misinterpreting the spec?

taylor-bsg, 2021-06-10:
The use case is any library where you want to instantiate a module that is instantiated by another module in the library that you are not using...

wsnyder, 2021-06-12:
Perhaps you or someone would be willing to make a pull request to support this?

Basically would require making tests, perhaps a basic one and another with module "faketop" instantiating "under" and "top".  "top" then instantiates "under" also, then use --top-module top.  The reason to have "under" this way is it will test the sorting properly considers only "top" not "faketop"

Then changing the code that sorts by level to start at the appropriate top, ignoring the "other" tops.

dpetrisko, 2021-07-08:
Fixed with #3026

closed 2021-07-08 by dpetrisko