verilator/verilator #3016: Can we remove scoped dpi export declarations?

gezalore, 2021-06-12:
The standard specifies that DPI exportd entry poitns are global functions, but we implement them as static functions on the top level:

```
int Vt_dpi_export::dpix_f_int(int i) {
    <implementation>
}
```

We then emmit a global function that simply calls the static.

```
#ifndef VL_DPIDECL_dpix_f_int_
#define VL_DPIDECL_dpix_f_int_
int dpix_f_int(int i) {
    // DPI export at t/t_dpi_export.v:43:25
    return Vt_dpi_export::dpix_f_int(i);
}
#endif
```

While I dislike the global namespace, and the statics are strictly in the "internals" section in the generated module header, I wonder if we can drop the statics and emit the global directly, so the top module header does not need to change when adding/removing DPI exports?

I.e.: Just emit this instead of the 2 functions above:

```
extern "C" int dpix_f_int(int i) {
    <implementation>
}
```

I'm raising the question because that is an interface change that might be noticeable, so if the intention was to indeed provide the static methods to be called by the user, then we might not want to do this.

wsnyder, 2021-06-12:
The reason it was designed this way was so that if there are multiple models the user is creating with different DPI exports of the same name, the user can choose to not use the extern versions (which would conflict), and instead call the class versions which disambiguates the overlap.  I don't know if this is being used though.

gezalore, 2021-06-12:
Hmm, thanks that makes sense. I will leave them for now and live with a rebuild when adding a new DPI export signature (hopefuly rare event).

closed 2021-06-12 by gezalore

gezalore, 2021-06-30:
Note as of #3036 these scoped dpi exports are part of the model class and no longer introduce additional header dependencies.