verilator/verilator #3018: 4.204 does not build on 32-bit platforms

tsailer, 2021-06-13:
g++ -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m32 -march=i686 -mtune=generic -msse2 -mfpmath=sse -mstackrealign -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection  -MMD -I. -I.. -I.. -I../../include -I../../include -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m32 -march=i686 -mtune=generic -msse2 -mfpmath=sse -mstackrealign -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/share/verilator\" -c ../Verilator.cpp -o Verilator.o
In file included from ../V3Number.h:24,
                 from ../V3Ast.h:25,
                 from ../Verilator.cpp:18:
../V3Hash.h:37:14: error: 'V3Hash::V3Hash(size_t)' cannot be overloaded with 'V3Hash::V3Hash(uint32_t)'
   37 |     explicit V3Hash(size_t val)
      |              ^~~~~~
../V3Hash.h:33:14: note: previous declaration 'V3Hash::V3Hash(uint32_t)'
   33 |     explicit V3Hash(uint32_t val)
      |              ^~~~~~
../V3Hash.h:56:12: error: 'V3Hash V3Hash::operator+(size_t) const' cannot be overloaded with 'V3Hash V3Hash::operator+(uint32_t) const'
   56 |     V3Hash operator+(size_t value) const { return *this + V3Hash(value); }
      |            ^~~~~~~~
../V3Hash.h:54:12: note: previous declaration 'V3Hash V3Hash::operator+(uint32_t) const'
   54 |     V3Hash operator+(uint32_t value) const { return *this + V3Hash(value); }
      |            ^~~~~~~~
../V3Hash.h:63:13: error: 'V3Hash& V3Hash::operator+=(size_t)' cannot be overloaded with 'V3Hash& V3Hash::operator+=(uint32_t)'
   63 |     V3Hash& operator+=(size_t value) { return *this += V3Hash(value); }
      |             ^~~~~~~~
../V3Hash.h:61:13: note: previous declaration 'V3Hash& V3Hash::operator+=(uint32_t)'
   61 |     V3Hash& operator+=(uint32_t value) { return *this += V3Hash(value); }
      |             ^~~~~~~~

On 32-bit platforms, size_t and uint32_t have the same size (and signedness)

gezalore, 2021-06-13:
Thanks for reporting this, I will take care of it.

wsnyder, 2021-06-13:
@gezalore thanks - perhaps just have a vluint64_t instead of size_t.

gezalore, 2021-06-14:
I have now fixed this (and other 32-bit build issues). I have also resurrected the `-m32` CI build, so hopefully the future will be brighter.

closed 2021-06-14 by gezalore

tsailer, 2021-06-14:
@gezalore Thank you!