verilator/verilator #3019: Tarball for Verilator?  And 'make test' fails (Fedora 32)

RichardMyersGV, 2021-06-13:
Hi,

It has been increasingly difficult (for me) to find a Verilator tarball of late.  It seems to get more and more 'hidden' or deprecated as time goes on.  Now I can't find it at all.  

Git doesn't fit well with my methods, and is currently borked in some way.  A tarball always works and I can go back and forth between versions more easily. 

Is there a tarball available somewhere?  Sorry if it's actually 'hidden' in plain site... I really have tried to find it multiple times. :)

Also, 'make test' hasn't worked for me since 4.202...

... (lots like this)
/usr/bin/ld: Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALL.o): in function `Vt_a1_first_cc::trace(VerilatedVcdC*, int, int)':
Vt_a1_first_cc__ALL.cpp:(.text+0x888): multiple definition of `Vt_a1_first_cc::trace(VerilatedVcdC*, int, int)'; Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALLsup.o):Vt_a1_first_cc__ALLsup.cpp:(.text+0x1f2): first defined here
/usr/bin/ld: Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALLsup.o): in function `Vt_a1_first_cc__Syms::Vt_a1_first_cc__Syms(Vt_a1_first_cc*, char const*)':
Vt_a1_first_cc__ALLsup.cpp:(.text+0x189): undefined reference to `VerilatedSyms::VerilatedSyms()'
/usr/bin/ld: Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALLsup.o): in function `Vt_a1_first_cc::trace(VerilatedVcdC*, int, int)':
Vt_a1_first_cc__ALLsup.cpp:(.text+0x22e): undefined reference to `VerilatedVcd::addCallback(void (*)(VerilatedVcd*, void*, unsigned int), void (*)(VerilatedVcd*, void*, unsigned int), void (*)(VerilatedVcd*, void*, unsigned int), void*)'
/usr/bin/ld: Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALLsup.o): in function `Vt_a1_first_cc::traceInitThis(Vt_a1_first_cc__Syms*, VerilatedVcd*, unsigned int)':
Vt_a1_first_cc__ALLsup.cpp:(.text+0x361): undefined reference to `VerilatedVcd::module(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: Vt_a1_first_cc__ALL.a(Vt_a1_first_cc__ALLsup.o): in function `Verilated::calcUnusedSigs()':
Vt_a1_first_cc__ALLsup.cpp:(.text._ZN9Verilated14calcUnusedSigsEv[_ZN9Verilated14calcUnusedSigsEv]+0x7): undefined reference to `Verilated::s_s'
collect2: error: ld returned 1 exit status
make[1]: *** [Vt_a1_first_cc.mk:63: Vt_a1_first_cc] Error 1
...
======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of make failed: g++  -I.  -MMD -I/home/rmm/src/verilator/test_regress/../include -I/home/rmm/src/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:07
make: *** [Makefile:257: smoke-test] Error 10

Thanks,
Richard

gezalore, 2021-06-14:
Regarding the error:

> Also, 'make test' hasn't worked for me since 4.202...

Those errors seem to be from a much older verilated.mk when the `__ALLsup.cpp` stil existed. That went away in  0f617988d45b5f7e8b72d30d3f2f6469a4c2ec7f which was In April 2020, so your are probably mixing versions, maybe VERILATOR_ROOT points to an old version?

RichardMyersGV, 2021-06-14:
Thanks!  I blew away the verilator subdirectory and did a:
git clone https://github.com/verilator/verilator   # Only first time
again.  It works now.  Perhaps there what something keeping 'git pull' from updating the directory.  I did 'unset VERILATOR_ROOT' .  Anyway, problem solved.  

I still would like the option of a tarball distribution...  It would save me some time.

Thanks,
Richard

wsnyder, 2021-06-14:
Glad it works.  Only two people have noticed the tarballs stopped being provided, so in the interest of putting effort where it benefits the majority, at this time I'm not planning on providing them.  Sorry.

closed 2021-06-14 by wsnyder