verilator/verilator #3020: Why do we have both WData and EData

gezalore, 2021-06-13:
Observe:
https://github.com/verilator/verilator/blob/13ddd0bc1cb598746dbfb24d5eb21973e1f3e0fa/include/verilated.h#L95-L106

I understand the purpose of CData, SData, IData and QData. I believe EData is one word within a VlWide, i.e.: anything wider than 64 bits is composed of EData.

I cannot see why we need WData? It is used all over the place, but why can't we just use EData instead?

Asking as the -m32 build I'm trying to fix is choking on an ambiguity between overloads of operators returning pointer vs EData and WData references.

wsnyder, 2021-06-13:
Anything wider than 64 bits is called a WData which is composed of EData.

gezalore, 2021-06-13:
I guess my confusion comes from stuff like this where we declare arrays of WData (as if there were VL_WQ_WORDS_E WData things, while in reality there is only one..):
https://github.com/verilator/verilator/blob/13ddd0bc1cb598746dbfb24d5eb21973e1f3e0fa/include/verilated.cpp#L504-L509
Based on what you said, sounds like one should only ever have a poitner to a WData, or instances of VLWide? It's a not really important, it's just late over here..

closed 2021-06-14 by gezalore

wsnyder, 2021-06-14:
An EData is always a single element so would never be arrayed, once it's arrayed it's wide data (WData).  Not sure this differentiation helps much, it's partially historical as EData was introduced later when I played with WDatas composed of 64 bit EDatas.

gezalore, 2021-06-14:
> An EData is always a single element so would never be arrayed

OK, let's add to the confusion then :confused::
https://github.com/verilator/verilator/blob/208f1504fbee321c059ad2b55271ab4f0757021a/include/verilated_heavy.h#L104-L108

I presume these days one could also just write this, wherever a WData array is written:
```
     VLWide<VL_WQ_WORDS_E> rhsw; 
```
IIRC standard C++ still only allows arrays of compile time known sizes, so this should always be possible.

wsnyder, 2021-06-14:
Yes, VlWide is the newest in the family and probably what we should be using instead of "WData x[]".  I still have a fear that the compile time or something will explode if we overuse VlWide, but so far it seems ok.