verilator/verilator #3023: Match on vector incorrectly transformed

pawel256, 2021-06-16:
version:
Verilator 4.204 2021-06-12 rev v4.204-18-gfa3c072

Please find attached a test in test_regress format.
[v4.204_bug.zip](https://github.com/verilator/verilator/files/6662991/v4.204_bug.zip)

When following SV code is converted into C++, the condition for the first match1_o is correct, but the condition for the match2_o is wrong.
(The only difference between match1_o and match2_o is the order of conditional expressions inside _if_ statement.)

```
module match
(
   input  logic [15:0] vec_i,
   input  logic b8_i,
   input  logic b12_i,
   output logic match1_o,
   output logic match2_o
);

always_comb
begin
   match1_o = 1'b0;
   if ( (vec_i[1:0] == 2'b0) &&
        (vec_i[4] == 1'b0) &&
        (vec_i[8] == b8_i) &&
        (vec_i[12] == b12_i)  )
   begin
      match1_o = 1'b1;
   end
end

always_comb
begin
   match2_o = 1'b0;
   if ( (vec_i[1:0] == 2'b0) &&
        (vec_i[8] == b8_i) &&
        (vec_i[12] == b12_i) &&
       (vec_i[4] == 1'b0)  )
   begin
      match2_o = 1'b1;
   end
end

endmodule
```

This is fine:
```
   vlTOPp->match1_o = 0U;
    if ((((IData)((0U == (0x13U & (IData)(vlTOPp->vec_i))))
          & ((1U & ((IData)(vlTOPp->vec_i) >> 8U))
             == (IData)(vlTOPp->b8_i))) & ((1U & ((IData)(vlTOPp->vec_i)
                                                  >> 0xcU))
                                           == (IData)(vlTOPp->b12_i)))) {
        vlTOPp->match1_o = 1U;
    }
```

But here is a problem:

```
    vlTOPp->match2_o = 0U;
    if ((IData)(((0x1100U == (0x1113U & (IData)(vlTOPp->vec_i)))
                 & (((1U & ((IData)(vlTOPp->vec_i)
                            >> 8U)) == (IData)(vlTOPp->b8_i))
                    & ((1U & ((IData)(vlTOPp->vec_i)
                              >> 0xcU)) == (IData)(vlTOPp->b12_i)))))) {
        vlTOPp->match2_o = 1U;
    }
```

`(0x1100U == (0x1113U & (IData)(vlTOPp->vec_i)))` condition matches only if `vec_i[8]` and `vec_i[12]` are both set to **1**
(The correct condition should be `(0U == (0x13U & (IData)(vlTOPp->vec_i)))` just like in match1_o code)

As far as I can find from the debug logs, the superfluous bits in the mask and the expected value seems to have been set by matchBitOpTree transformation in V3Const.cpp
(the tree before this transformation seems fine to me)

```
- V3Const.cpp:731:    Transformed leaf of bit tree to CCAST 0x276fec0 <e2515#> {c36ah} @dt=0x2770730@(G/wu32/1) sz32
Call matchBitOpTree[1]
     AND 0x27806c0 <e2300#> {c36ah} @dt=0x2770730@(G/wu32/1)
    1: AND 0x2780780 <e2295#> {c34ah} @dt=0x2770730@(G/wu32/1)
    1:1: AND 0x2780840 <e2276#> {c32ah} @dt=0x2770730@(G/wu32/1)
    1:1:1: EQ 0x2780900 <e2244#> {c31at} @dt=0x2770730@(G/wu32/1)
    1:1:1:1: CONST 0x27809c0 <e2240#> {c31aw} @dt=0x2776dc0@(G/wu32/2)  2'h0
    1:1:1:2: AND 0x276daa0 <e2257#> {c31an} @dt=0x2776dc0@(G/wu32/2)
    1:1:1:2:1: CONST 0x2779340 <e2253#> {c31an} @dt=0x277ac90@(G/w32)  32'h3
    1:1:1:2:2: VARREF 0x2780c00 <e2484#> {c31ai} @dt=0x2776dc0@(G/wu32/2)  vec_i [RV] <- VAR 0x277c4a0 <e976#> {c3ay} @dt=0x276db80@(G/w16)  vec_i [PI] INPUT [P] [VSTATIC]  PORT
    2: NOT 0x2781e00 <e2299#> {c37ar} @dt=0x2770730@(G/wu32/1)
    2:1: SHIFTR 0x2776b00 <e2436#> {c37an} @dt=0x2770730@(G/wu32/1)
    2:1:1: VARREF 0x2781f90 <e2426#> {c37ai} @dt=0x27718e0@(G/wu32/16)  vec_i [RV] <- VAR 0x277c4a0 <e976#> {c3ay} @dt=0x276db80@(G/w16)  vec_i [PI] INPUT [P] [VSTATIC]  PORT
    2:1:2: CONST 0x27820d0 <e2427#> {c37ao} @dt=0x276f8c0@(G/swu32/4)  4'h4

Result:
     CCAST 0x276fec0 <e2515#> {c36ah} @dt=0x2770730@(G/wu32/1) sz32
    1: AND 0x276f670 <e2514#> {c36ah} @dt=0x2778b40@(G/nw1)
    1:1: EQ 0x276f100 <e2509#> {c31ai} @dt=0x2778b40@(G/nw1)
    1:1:1: CONST 0x27833e0 <e2499#> {c31ai} @dt=0x277ac90@(G/w32)  32'h1100
    1:1:2: AND 0x2774db0 <e2500#> {c31ai} @dt=0x277ac90@(G/w32)
    1:1:2:1: CONST 0x2782e30 <e2495#> {c31ai} @dt=0x277ac90@(G/w32)  32'h1113
    1:1:2:2: VARREF 0x277c600 <e2496#> {c31ai} @dt=0x2776dc0@(G/wu32/2)  vec_i [RV] <- VAR 0x277c4a0 <e976#> {c3ay} @dt=0x276db80@(G/w16)  vec_i [PI] INPUT [P] [VSTATIC]  PORT
    1:2: AND 0x277b7d0 <e2510#> {c36ah} @dt=0x2770730@(G/wu32/1)
    1:2:1: EQ 0x2781020 <e2505#> {c33ar} @dt=0x2770730@(G/wu32/1)
    1:2:1:1: AND 0x2779dc0 <e2275#> {c33an} @dt=0x2770730@(G/wu32/1)
    1:2:1:1:1: CONST 0x2778510 <e2271#> {c33an} @dt=0x277ac90@(G/w32)  32'h1
    1:2:1:1:2: SHIFTR 0x2778ee0 <e2402#> {c33an} @dt=0x2770730@(G/wu32/1)
    1:2:1:1:2:1: VARREF 0x27811b0 <e2392#> {c33ai} @dt=0x27718e0@(G/wu32/16)  vec_i [RV] <- VAR 0x277c4a0 <e976#> {c3ay} @dt=0x276db80@(G/w16)  vec_i [PI] INPUT [P] [VSTATIC]  PORT
    1:2:1:1:2:2: CONST 0x27812f0 <e2393#> {c33ao} @dt=0x276f8c0@(G/swu32/4)  4'h8
    1:2:1:2: VARREF 0x27815d0 <e2261#> {c33au} @dt=0x2770730@(G/wu32/1)  b8_i [RV] <- VAR 0x277c740 <e981#> {c4ar} @dt=0x276d390@(G/w1)  b8_i [PI] INPUT [P] [VSTATIC]  PORT
    1:2:2: EQ 0x2781710 <e2506#> {c35as} @dt=0x2770730@(G/wu32/1)
    1:2:2:1: AND 0x276ec50 <e2294#> {c35an} @dt=0x2770730@(G/wu32/1)
    1:2:2:1:1: CONST 0x276dc60 <e2290#> {c35an} @dt=0x277ac90@(G/w32)  32'h1
    1:2:2:1:2: SHIFTR 0x277a420 <e2419#> {c35an} @dt=0x2770730@(G/wu32/1)
    1:2:2:1:2:1: VARREF 0x27818a0 <e2409#> {c35ai} @dt=0x27718e0@(G/wu32/16)  vec_i [RV] <- VAR 0x277c4a0 <e976#> {c3ay} @dt=0x276db80@(G/w16)  vec_i [PI] INPUT [P] [VSTATIC]  PORT
    1:2:2:1:2:2: CONST 0x27819e0 <e2410#> {c35ao} @dt=0x276f8c0@(G/swu32/4)  4'hc
    1:2:2:2: VARREF 0x2781cc0 <e2280#> {c35av} @dt=0x2770730@(G/wu32/1)  b12_i [RV] <- VAR 0x277cae0 <e987#> {c5ar} @dt=0x276d390@(G/w1)  b12_i [PI] INPUT [P] [VSTATIC]  PORT
- V3Const__gen.cpp:2862:0x27806c0 TREEOPC( AstAnd nodep->widthMin() == 1, matchBitOpTree(nodep) , DONE )
```

Thanks,
Pawel

pawel256, 2021-06-16:
test passes when `V3Const.cpp` is reverted to version 4351abfe7107487a524fe51b631119487c2fa6b9,
with the following code generated:
```
    if (((((0U == (3U & (IData)(vlTOPp->t__DOT__vec_i)))
           & ((1U & ((IData)(vlTOPp->t__DOT__vec_i)
                     >> 8U)) == (IData)(vlTOPp->t__DOT__b8_i)))
          & ((1U & ((IData)(vlTOPp->t__DOT__vec_i)
                    >> 0xcU)) == (IData)(vlTOPp->t__DOT__b12_i)))
         & (~ ((IData)(vlTOPp->t__DOT__vec_i) >> 4U)))) {
        vlTOPp->t__DOT__match2_o = 1U;
    }
```
(no merge of matches vec_i[1:0] and vec_i[4])

and fails after the next change to `V3Const.cpp` is included (version 8624ce6a84dcd0908a9d721420e1f790ffaf4ba1)

wsnyder, 2021-06-17:
@yTakatsukasa can you look please, thanks.

yTakatsukasa, 2021-06-17:
Thank you for the standard format test.
I confirmed that the test fails, but adding `-Oo` option that disables bittree opt make the test pass.

`-Oo` option can be used as a workaround until I fix this bug.

yTakatsukasa, 2021-06-17:
I think I understand the root cause and have an initial version of the fix that passes your example and regression tests.
I will polish it and send PR hopefully within a few days.

pawel256, 2021-06-17:
Thank you very much for quick response.

yTakatsukasa, 2021-06-20:
#3030 should fix this issue. I'd appreciate if you could try your original design with the latest master. Thanks!

closed 2021-06-20 by yTakatsukasa

pawel256, 2021-06-23:
I am glad to confirm that after #3030 my regressions completed with no problem
Thank you very much
Pawel

PR #3030 (2021-06-20): Fiix incorrect result by bit tree opt (#3023)
Fixes #3023

The root cause is some member variables are not properly restored to the original value when an AST subtree cannot be optimized.

I will push a commit that adds test and see CI fails, then push a fix.

ext tests pass too.