verilator/verilator #3028: Unsupported: 4-state numbers in this context

msalvi123, 2021-06-17:
A function within a package that drives a default of 'x causes the error.

function automatic logic [15:0] copy_mask;
      input [15:0]  mask_in;
      input [1:0]   vsew;
      case(vsew[1:0])
	2'b00:                         copy_mask[15:0] =  mask_in[15:0];
	2'b01:   for(int i=0;i<16;i++) copy_mask[i]    =  mask_in[i[3:1]];
	2'b10:   for(int i=0;i<16;i++) copy_mask[i]    =  mask_in[i[3:2]];
	default:                       copy_mask[15:0] = 'x;
      endcase
   endfunction

%Error-UNSUPPORTED: : Unsupported: 4-state numbers in this context
   57 |  default: copy_mask[15:0] = 'x;

wsnyder, 2021-06-17:
This works standalone and is probably caused by how the calling code is using it.  Can you please show a standalone example that causes this error?

msalvi123, 2021-06-18:
The function is defined in a package, pulled it in this module to create a test case.

module test(input [1:0] size,
	    input [15:0] mask,
	    output logic [15:0] sized_mask);
   
   function automatic logic [15:0] copy_mask;
      input [15:0]  mask_in;
      input [1:0]   vsew;
      case(vsew[1:0])
	2'b00:                         copy_mask[15:0] =  mask_in[15:0];
	2'b01:   for(int i=0;i<16;i++) copy_mask[i]    =  mask_in[i[3:1]];
	2'b10:   for(int i=0;i<16;i++) copy_mask[i]    =  mask_in[i[3:2]];
	default:                       copy_mask[15:0] = 'x;
      endcase
   endfunction
   
 
   **assign sized_mask[15:0] = copy_mask(mask, size[1:0]);**

endmodule

wsnyder, 2021-06-18:
This file verilates OK for me.  Perhaps you need to use a newer version of Verilator?

msalvi123, 2021-06-18:
Version installed : Verilator 4.040 2020-08-15 rev v4.016-1103-g28cf0740

wsnyder, 2021-06-18:
Please try the git master version.

msalvi123, 2021-06-20:
It appears that the issue is with the function declaration within a package.
When the function is pulled out of a package, the code Verilates OK, with the installed version.

wsnyder, 2021-08-24:
Seems git master is ok then, if I misunderstood you have a failing case on git master please post an example and reopen.

closed 2021-08-24 by wsnyder