verilator/verilator #3029: 4.204 build failure compiling V3Hash.cpp

margej, 2021-06-18:
```
g++ -std=gnu++11  -MMD -I. -I.. -I.. -I../../include -I../../include -O2 -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Hash.cpp -o V3Hash.o

../V3Hash.cpp: In function ‘std::ostream& operator<<(std::ostream&, const V3Hash&)’:
../V3Hash.cpp:26:15: error: no match for ‘operator<<’ (operand types are ‘std::ostream {aka std::basic_ostream<char>}’ and ‘std::ios_base&(std::ios_base&)’)
     return os << std::hex << std::setw(8) << std::setfill('0') << rhs.value();
               ^
../V3Hash.cpp:26:15: note: candidates are:
../V3Hash.cpp:25:15: note: std::ostream& operator<<(std::ostream&, const V3Hash&)
 std::ostream& operator<<(std::ostream& os, const V3Hash& rhs) {
               ^
../V3Hash.cpp:25:15: note:   no known conversion for argument 2 from ‘std::ios_base&(std::ios_base&)’ to ‘const V3Hash&’
In file included from ../V3Hash.cpp:20:0:
/usr/include/c++/4.8.2/iomanip:308:5: note: template<class _CharT, class _Traits, class _MoneyT> std::basic_ostream<_CharT, _Traits>& std::operator<<(std::basic_ostream<_CharT, _Traits>&, std::_Put_money<_MoneyT>)
     operator<<(basic_ostream<_CharT, _Traits>& __os, _Put_money<_MoneyT> __f)
     ^
```

4.202 builds just fine. Is this due to some old packages? I'm running CentOS 7

```
gcc version 4.8.5 20150623 (Red Hat 4.8.5-39) (GCC) 
```

margej, 2021-06-18:
NOTE: This is a 64-bit system, so this is not #3018

gezalore, 2021-06-18:
> NOTE: This is a 64-bit system, so this is not #3018

Hah, you beat me to it.

Could you check if `#include <ios>` in `V3Hash.cpp` fixes this please?

margej, 2021-06-21:
#include <ios> did NOT fix the issue, but
#include <iostream> did

Thanks

gezalore, 2021-06-21:
Thanks for confirming. This is now fixed on master.

commit 7ecc2d4d (2021-06-21): V3Hash: Add missing include.

Fixes #3029
closed 2021-06-21 by gezalore