verilator/verilator #3031: Intel macOS Big Sur 11.4 make test fails: "ld: file too small (length=8) file 'Vt_a1_first_cc__ALL.a' for architecture x86_64"

badwin00, 2021-06-19:
Been trying to install for a few days, i've `brew install` all the dependencies i can. Getting Verilator itself through brew throws the same linker error when i try to use it. It works through docker but produces linux binaries, which is not ideal for me. Any guidance would be appreciated, thank you.

<div>
Samuels-MBP :: ~/Tools/verilator ‹master*› % ./bin/verilator --version                                                       2 ↵
Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)
Samuels-MBP :: ~/Tools/verilator ‹master*› % make test                                                                          
------------------------------------------------------------
making verilator in src
/Library/Developer/CommandLineTools/usr/bin/make -C src 
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Nothing to be done for `serial'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Nothing to be done for `serial_vlcov'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
/Library/Developer/CommandLineTools/usr/bin/make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Nothing to be done for `serial'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
test_regress/t/t_a1_first_cc.pl
======================================================================
dist/t_a1_first_cc: ==================================================
-Skip: dist/t_a1_first_cc: scenario 'dist' not enabled for test
dist/t_a1_first_cc: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
vlt/t_a1_first_cc: ==================================================
	perl ../bin/verilator --debug --debugi 0 --gdbbt --no-dump-tree -V
"0x7ffeef22b4f0s": not in executable format: file format not recognized
warning: ~/SandBox/homecvs/v4/verilator/src/.gdbinit: No such file or directory
No executable file specified.
Use the "file" or "exec-file" command.
No stack.
-Info: --gdbbt ignored: gdb doesn't seem to be working
Starting Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)
Starting Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)
Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)

Copyright 2003-2021 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /Users/sambadwin/Tools/verilator
    SystemC system-wide = 0

Environment:
    MAKE               = make
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /Users/sambadwin/Tools/verilator
    VERILATOR_BIN      = 

Features (based on environment or compiled-in support):
    SystemC found      = 0
	perl /Users/sambadwin/Tools/verilator/bin/verilator --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc -OD --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc t/t_a1_first_cc.v    > obj_vlt/t_a1_first_cc/vlt_compile.log
"0x7ffee56974f0s": not in executable format: file format not recognized
[... 3 lines cut ...]
No stack.
-Info: --gdbbt ignored: gdb doesn't seem to be working
Starting Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)
Starting Verilator 4.205 devel rev v4.204-35-g6c9c16c3 (mod)
	make -C obj_vlt/t_a1_first_cc -f /Users/sambadwin/Tools/verilator/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_a1_first_cc TEST_OBJ_DIR=obj_vlt/t_a1_first_cc CPPFLAGS_DRIVER=-DT_A1_FIRST_CC  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_a1_first_cc    > obj_vlt/t_a1_first_cc/vlt_gcc.log
driver: Entering directory '/Users/sambadwin/Tools/verilator/test_regress/obj_vlt/t_a1_first_cc'
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
g++    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o Vt_a1_first_cc__ALL.a      -o Vt_a1_first_cc
ld: file too small (length=8) file 'Vt_a1_first_cc__ALL.a' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vt_a1_first_cc] Error 1
driver: Leaving directory '/Users/sambadwin/Tools/verilator/test_regress/obj_vlt/t_a1_first_cc'
%Warning: vlt/t_a1_first_cc: Exec of make failed: ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp

vlt/t_a1_first_cc: %Error: Exec of make failed: ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
vlt/t_a1_first_cc: FAILED: Exec of make failed: ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new [... cut ...] quality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:02
make: *** [smoke-test] Error 10
</div>

stephenry, 2021-06-19:
This is related to #2999.

badwin00, 2021-06-19:
progress, i've done `brew install binutils` and now feed it to configure using `./configure AR=/usr/local/opt/binutils/bin/ar`

running make test after that gives me:

Samuels-MBP :: ~/Tools/verilator ‹master*› % make test                                                                          
------------------------------------------------------------
making verilator in src
/Library/Developer/CommandLineTools/usr/bin/make -C src 
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Nothing to be done for `serial'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Nothing to be done for `serial_vlcov'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
/Library/Developer/CommandLineTools/usr/bin/make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Nothing to be done for `serial'.
/Library/Developer/CommandLineTools/usr/bin/make -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
      Compile flags:  g++ -std=gnu++14 -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/Users/sambadwin/Tools/verilator"
test_regress/t/t_a1_first_cc.pl
======================================================================
dist/t_a1_first_cc: ==================================================
-Skip: dist/t_a1_first_cc: scenario 'dist' not enabled for test
dist/t_a1_first_cc: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Unsup 0  Time 0:00
======================================================================
vlt/t_a1_first_cc: ==================================================
	perl ../bin/verilator --debug --debugi 0 --gdbbt --no-dump-tree -V
"0x7ffee0da04f0s": not in executable format: file format not recognized
warning: ~/SandBox/homecvs/v4/verilator/src/.gdbinit: No such file or directory
No executable file specified.
Use the "file" or "exec-file" command.
No stack.
-Info: --gdbbt ignored: gdb doesn't seem to be working
Starting Verilator 4.205 devel rev v4.204-38-geebda248 (mod)
Starting Verilator 4.205 devel rev v4.204-38-geebda248 (mod)
Verilator 4.205 devel rev v4.204-38-geebda248 (mod)

Copyright 2003-2021 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /Users/sambadwin/Tools/verilator
    SystemC system-wide = 0

Environment:
    MAKE               = make
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /Users/sambadwin/Tools/verilator
    VERILATOR_BIN      = 

Features (based on environment or compiled-in support):
    SystemC found      = 0
	perl /Users/sambadwin/Tools/verilator/bin/verilator --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc -OD --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc t/t_a1_first_cc.v    > obj_vlt/t_a1_first_cc/vlt_compile.log
"0x7ffee32804f0s": not in executable format: file format not recognized
warning: ~/SandBox/homecvs/v4/verilator/src/.gdbinit: No such file or directory
[... 15 lines cut ...]
      _main in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::trace(VerilatedVcdC*, int, int)", referenced from:
      _main in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::eval_step()", referenced from:
      Vt_a1_first_cc::eval() in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::Vt_a1_first_cc(VerilatedContext*, char const*)", referenced from:
      Vt_a1_first_cc::Vt_a1_first_cc(char const*) in Vt_a1_first_cc__main.o
  "Vt_a1_first_cc::~Vt_a1_first_cc()", referenced from:
[... cut ...] heses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:01
make: *** [smoke-test] Error 10

wsnyder, 2021-07-07:
Try editing the test example Makefile to define VL_NO_LEGACY  (-DVL_NO_LEGACY).  I suspect macOS doesn't support weak symbols.

badwin00, 2021-07-08:
No luck, same issue. Command used:
$(VERILATOR) -cc --exe -DVL_NO_LEGACY --build -j top.v sim_    main.cpp

<pre>
-- Verilator hello-world simple example
-- VERILATE & BUILD --------
/Users/sambadwin/Tools/verilator/bin/verilator -cc --exe -DVL_NO_LEGACY --build -j top.v sim_main.cpp
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o sim_main.o ../sim_main.cpp
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated.o /Users/sambadwin/Tools/verilator/include/verilated.cpp
/usr/local/bin/perl /Users/sambadwin/Tools/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root.cpp Vtop___024root__Slow.cpp Vtop__Syms.cpp > Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
Archive /usr/local/opt/binutils/bin/ar -rcs Vtop__ALL.a Vtop__ALL.o
g++    sim_main.o verilated.o Vtop__ALL.a      -o Vtop
ld: warning: ignoring file Vtop__ALL.a, building for macOS-x86_64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture x86_64:
  "Vtop::final()", referenced from:
      _main in sim_main.o
  "Vtop::eval_step()", referenced from:
      _main in sim_main.o
  "Vtop::Vtop(char const*)", referenced from:
      _main in sim_main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vtop] Error 1
rm Vtop__ALL.verilator_deplist.tmp
%Error: make -C obj_dir -f Vtop.mk -j exited with 2
%Error: Command Failed /Users/sambadwin/Tools/verilator/bin/verilator_bin -cc --exe -DVL_NO_LEGACY --build -j top.v sim_main.cpp
make: *** [default] Error 2
</pre>

wsnyder, 2021-07-08:
Sorry I should have been clearer, that's a compiler not verilator flag, so "-CFLAGS -DVL_NO_LEGACY".

badwin00, 2021-07-08:
Looks like progress, after running Make in examples/make_hello_c with `$(VERILATOR) -cc --exe --build -j top.v sim_main.cpp -CFLAGS -DVL_NO_LEGACY` I now get: 
<pre>
-- Verilator hello-world simple example
-- VERILATE & BUILD --------
/Users/sambadwin/Tools/verilator/bin/verilator -cc --exe --build -j top.v sim_main.cpp -CFLAGS -DVL_NO_LEGACY 
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_NO_LEGACY  -std=gnu++14 -Os -c -o sim_main.o ../sim_main.cpp
/usr/local/bin/perl /Users/sambadwin/Tools/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root.cpp Vtop___024root__Slow.cpp Vtop__Syms.cpp > Vtop__ALL.cpp
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_NO_LEGACY  -std=gnu++14 -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive /usr/local/opt/binutils/bin/ar -rcs Vtop__ALL.a Vtop__ALL.o
../sim_main.cpp:28:24: error: no member named 'gotFinish' in 'Verilated'
    while (!Verilated::gotFinish()) {
            ~~~~~~~~~~~^
1 error generated.
make[1]: *** [sim_main.o] Error 1
rm Vtop__ALL.verilator_deplist.tmp
%Error: make -C obj_dir -f Vtop.mk -j exited with 2
%Error: Command Failed /Users/sambadwin/Tools/verilator/bin/verilator_bin -cc --exe --build -j top.v sim_main.cpp -CFLAGS -DVL_NO_LEGACY
make: *** [default] Error 2
</pre>
after trying to fix this error myself by replacing Verilated with VerilatedContext->gotFinish() I now get:
<pre>
-- Verilator hello-world simple example
-- VERILATE & BUILD --------
/Users/sambadwin/Tools/verilator/bin/verilator -cc --exe --build -j top.v sim_main2.cpp -CFLAGS -DVL_NO_LEGACY 
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_NO_LEGACY  -std=gnu++14 -Os -c -o sim_main2.o ../sim_main2.cpp
/usr/local/bin/perl /Users/sambadwin/Tools/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root.cpp Vtop___024root__Slow.cpp Vtop__Syms.cpp > Vtop__ALL.cpp
ccache g++  -I.  -MMD -I/Users/sambadwin/Tools/verilator/include -I/Users/sambadwin/Tools/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_NO_LEGACY  -std=gnu++14 -Os -c -o Vtop__ALL.o Vtop__ALL.cpp
echo "" > Vtop__ALL.verilator_deplist.tmp
Archive /usr/local/opt/binutils/bin/ar -rcs Vtop__ALL.a Vtop__ALL.o
g++    sim_main2.o verilated.o Vtop__ALL.a      -o Vtop
ld: warning: ignoring file Vtop__ALL.a, building for macOS-x86_64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture x86_64:
  "Vtop::final()", referenced from:
      _main in sim_main2.o
  "Vtop::eval_step()", referenced from:
      _main in sim_main2.o
  "Vtop::Vtop(VerilatedContext*, char const*)", referenced from:
      _main in sim_main2.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vtop] Error 1
rm Vtop__ALL.verilator_deplist.tmp
%Error: make -C obj_dir -f Vtop.mk -j exited with 2
%Error: Command Failed /Users/sambadwin/Tools/verilator/bin/verilator_bin -cc --exe --build -j top.v sim_main2.cpp -CFLAGS -DVL_NO_LEGACY
make: *** [default] Error 2
</pre>

wsnyder, 2021-07-08:
Searching on the error "building for macOS-x86_64 but attempting to link with file built for macOS-x86_64" suggests an inconsistent tool chain, e.g. see https://github.com/bitcoin/bitcoin/issues/20825

badwin00, 2021-07-09:
hmm, any tips on how to verify whether or not my toolchain matches up? The only tool i'm going out of my way to swap out for GNU rather than Apple is `ar` for the reasons above. I haven't messed with my $PATH in any way that would modify the default tooling. 

Maybe these will help narrow it down?
config.log: https://pastebin.com/ff6ajpnG
brew list: https://pastebin.com/uPLijid7

wsnyder, 2021-07-10:
Don't know myself as I don't use MacOs, sorry.  I'd note MacOs is a continual source of pain as often the tools are broken when Apple ships them, so I wouldn't assume what comes with the OS is consistent/correct.

wsnyder, 2021-08-24:
Didn't hear of any Verilator change needed here - feel free to reopen if something needs to be done.

closed 2021-08-24 by wsnyder

gajaka, 2021-09-04:
I am on Macos Catalina 10.15.7 and getting the same issue with `(!Verilated::gotFinish())` any progress on this ?

badwin00, 2021-09-04:
> after trying to fix this error myself by replacing Verilated::gotFinish() with VerilatedContext->gotFinish() I now get:

@gajaka I mention my workaround above, using VerilatedContext instead of Verilated worked for me

gajaka, 2021-09-05:
I fixed using different approach and all examples ( excluding `cmake_protect_lib` ) works fine. 
I removed flags `-CFLAGS -DVL_NO_LEGACY` and binutils, then I have installed systemc by `cmake` and following documentation I have installed `verilator`. Your post was useful starting point. Thanks.

quark17, 2021-12-24:
> I suspect macOS doesn't support weak symbols.

You have to give `-undefined dynamic_lookup` to the linker.  The link errors go away if I add this to `test_regress/Makefile_obj`:
```
LDFLAGS += -undefined dynamic_lookup
```
Is there some way to make to this automatically incorporated for macOS?  (I don't yet know enough to understand what goes in the configure script, what goes in `verilator.mk`, what goes in Verilator's generated makefiles, etc.)

> ```
> "0x7ffee0da04f0s": not in executable format: file format not recognized
> warning: ~/SandBox/homecvs/v4/verilator/src/.gdbinit: No such file or directory
> No executable file specified.
> Use the "file" or "exec-file" command.
> No stack.
> ```

Is this something to be concerned about?  I too get these messages, although the executable name is different:
```
"/bin/echo": not in executable format: file format not recognized
warning: ~/SandBox/homecvs/v4/verilator/src/.gdbinit: No such file or directory
No executable file specified.
Use the "file" or "exec-file" command.
No stack.
```
These are the contents of the `vlt_compile.log` file.  The tests still run, but I'd like to see what's going on here and at least clean it up.  What generates this? where can I find what commands are being executed here?

> suggests an inconsistent tool chain

I suspect this comes from trying to replace macOS's `ar` with a GNU version, and thereby pulling in some GNU binutils but not fully switching away from macOS's utils.  I'd rather not replace `ar` and instead make Verilator support BSD `ar`.  I have submitted a PR for this, #3256.  With only that small change (instead of any of the workarounds mentioned on this thread), Verilator builds and runs fine; and the tests (with LDFLAGS updated for weak symbols) seem to run mostly fine.  Although there are some failures, to be looked at.  The first couple are because `make` is version 3.81.  Some might be tool/environment issues.

wsnyder, 2021-12-24:
>LDFLAGS += -undefined dynamic_lookup

Rather than do it based on an OS, probably test in configure.ac to see if that option is allowed by the linker and if, so, add it to LDFLAGS.