verilator/verilator #3032: Make top module part of Syms, emit new top wrapper in addition

gezalore, 2021-06-19:
In the emitted code, currently the top level module is special. Every cell, apart from the top is part of Syms, while the Top instantiates sysms and has a bunch of additional gubbins like the eval loops, thread pools, trace configuration routines, etc. This complicates quite a lot of the internals and I think we could do better by doing the following:
- Emit top level module as a regular module with name `topClassName() + "__TOP"` or similar
- Make the top level instance part of Syms just like any other cell.
- Emit a new wrapper with the current top module name `topClassName()` that instantiates Syms, and proxies the user accessible interface as needed.
- Emit the 'additional gubbins' mentioned above as part of this wrapper.

For compatibility, we need to be careful not to change the user interface of the `topClassName()` module which is now this newly created wrapper. Luckily this is relatively simple:
- Major methods like `eval` on the toplevel are part of the 'additional gubbins' anyway
- Public signals/parameters we can proxy via references:
```
VL_MODULE(Vtop) {
private:
  Vtop__Syms vlSyms;

public:
  VL_IN8(&clk,0,0);
  VL_IN8(&rst,0,0);

  Vtop();
}

Vtop::Vtop()
  : vlSyms(....)
  , clk(vlSyms.wherever.the.real.clk.is)
  , rst(vlSyms.wherever.the.real.rst.is)
 {}
```
- Any other public functions we can forward as appropriate.

This would have the following benefits:
- V3EmitC could be split in half, separating generation of the 'additional gubbins' into a V3EmitCWrapper (or similar), which is not needed for the common modules.
- There would be a fewer special cases for ->isTop() elsewhere.
- Re #3016, scoped DPI export declarations could move into the wrapper, removing a header dependency
- Further reduce header dependencies as model code should not care about the top wrapper gubbins.

All of the above would facility further work towards #2140/#2958

The only negative I can see is the extra reference indirection when accession top level IO from outside, that however really should not be critical for any model of non-trivial complexity.

One unknown is whether the refernce forwarding would work for SystemC IO (I don't know if SystemC cares at all). Worst case there is always the option of copy in and copy out before and after the eval loop...

How do you feel about doing this?

wsnyder, 2021-06-19:
There's the Verilator wrapped top, then there's the Verilog top module(s), these are not the same.  The Verilog top module is what --top-module is requesting, and this is "normal" in the sense that it is part of Syms.  The wrapper top has the I/O which are a union of the Verilog tops (made by  wrapTopCell).

This is perhaps somewhat historical (predating when Syms existed), but also SystemC must have the wrapper as the signals need to be declared in the header (not in Syms), as you somewhat guessed.  I don't see any way around having the top be different in that respect.

gezalore, 2021-06-19:
> There's the Verilator wrapped top, then there's the Verilog top module(s), these are not the same. The Verilog top module is what --top-module is requesting, and this is "normal" in the sense that it is part of Syms. 

That does not quite fit my mental model, but we might be talking about different things. The output of t_EXAMPLE.pl is this Syms file:

```
class Vt_EXAMPLE__Syms : public VerilatedSyms {
  public:

    // LOCAL STATE
    const char* __Vm_namep;
    bool __Vm_didInit;

    // SUBCELL STATE
    Vt_EXAMPLE*                    TOPp;

    // CREATORS
    Vt_EXAMPLE__Syms(VerilatedContext* contextp, Vt_EXAMPLE* topp, const char* namep);
    ~Vt_EXAMPLE__Syms();

    // METHODS
    inline const char* name() { return __Vm_namep; }

} VL_ATTR_ALIGNED(VL_CACHE_LINE_BYTES);
```
Note the above does not have a TOP instance, just a pointer to it. If there were other sub-cells, they would be *instances* (and not poitners), listed below `TOPp`.

The only other class emitted is the top level module:
```
VL_MODULE(Vt_EXAMPLE) {
  public:

    // PORTS
    VL_IN8(clk,0,0);

    // LOCAL SIGNALS
    IData/*31:0*/ t__DOT__cyc;
    IData/*31:0*/ t__DOT__out;
    QData/*63:0*/ t__DOT__crc;
    QData/*63:0*/ t__DOT__sum;

    // LOCAL VARIABLES
    CData/*0:0*/ __Vclklast__TOP__clk;

    // INTERNAL VARIABLES
    Vt_EXAMPLE__Syms* vlSymsp;  // Symbol table

    // CONSTRUCTORS
  private:
    VL_UNCOPYABLE(Vt_EXAMPLE);  ///< Copying not allowed
  public:
    Vt_EXAMPLE(VerilatedContext* contextp, const char* name = "TOP");
    Vt_EXAMPLE(const char* name = "TOP")
      : Vt_EXAMPLE(nullptr, name) {}
    ~Vt_EXAMPLE();

    // API METHODS
    /// Return current simulation context for this model.
    /// Used to get to e.g. simulation time via contextp()->time()
    VerilatedContext* contextp() const;
    /// Evaluate the model.  Application must call when inputs change.
    void eval() { eval_step(); }
    /// Evaluate when calling multiple units/models per time step.
    void eval_step();
    /// Evaluate at end of a timestep for tracing, when using eval_step().
    /// Application must call after all eval() and before time changes.
    void eval_end_step() {}
    /// Simulation complete, run final blocks.  Application must call on completion.
    void final();

    // INTERNAL METHODS
[... 101 lines cut ...]
    void eval() { eval_step(); }
    /// Evaluate when calling multiple units/models per time step.
    void eval_step();
    /// Evaluate at end of a timestep for tracing, when using eval_step().
    /// Application must call after all eval() and before time changes.
    void eval_end_step() {}
    /// Simulation complete, run final blocks.  Application must call on completion.
    void final();

   // NO internal methods
} VL_ATTR_ALIGNED(VL_CACHE_LINE_BYTES);
```

The user theninstantiates and iteracts with with this new wrapper class, that looks and feels like the old/current top.

For SystemC, are you saying this would not work?
```
SC_MODULE(Vt_a2_first_sc) {
  public:

    // PORTS
    sc_in<bool> &clk; // reference to the actual sc_in<bool> inside the instance of Vt_a2_first_sc__TOP living in the Syms owned by this class
...
}
```

gezalore, 2021-06-19:
Note in SC, only the new wrapper would be an SC_MODULE, and everything else including the top, would be a VL_MODULE like now.

gezalore, 2021-06-19:
BTW I don't have much idea about SystemC, but I am under the impression that it is just a library (i.e.: there are no special build tools that process sources to derive info the C compiler does not itself know), so I don't see how it could tell the difference between the port and a reference to it short of relying on type traits.

gezalore, 2021-06-19:
> There's the Verilator wrapped top, then there's the Verilog top module(s), these are not the same. The Verilog top module is what --top-module is requesting, and this is "normal" in the sense that [...]

OK, I think I'm with you, in most cases the actual top then get's inlined into that wrapper so it doesn't show up in the output separate. What I meant by 'top level' is whatever AstNodeModule that returns true for `isTop()` towards the backend of Verilator, and by 'wrapper' I was talking about something completely different (as I described above).

gezalore, 2021-06-20:
Did you have a change to think about this? I have hand prototyped t_a2_first_sc and it seems to work with the suggested reference based forwarding:

```
SC_MODULE(Vt_a2_first_sc) {
  public:
    Vt_a2_first_sc__Syms vlSyms;  // Symbol table
    sc_in<bool>& clk;
    ....
} VL_ATTR_ALIGNED
```

And in the constructor:

```
Vt_a2_first_sc::Vt_a2_first_sc(sc_module_name)
    : vlSyms(nullptr, name())
    , clk(vlSyms.TOP.clk)
 {
    std::cout << "constructor: " << &vlSyms.TOP.clk << std::endl;

    // Sensitivities on all clocks and combo inputs
    SC_METHOD(eval);
    sensitive << clk;
}
```

---

The Syms then look like:
```
class Vt_a2_first_sc__Syms : public VerilatedSyms {
   ...
    // SUBCELL STATE
    Vt_a2_first_sc__TOP                TOP;
   ...
}
```

And Vt_a2_first_sc__TOP is just a VL_MODULE:
```
VL_MODULE(Vt_a2_first_sc__TOP) {
  public:
    sc_in<bool> clk;

    CData/*0:0*/ __Vcellinp__t__clk;
    CData/*0:0*/ __Vclklast__TOP____Vcellinp__t__clk;

    Vt_a2_first_sc__Syms* vlSymsp;  // Symbol table

  ...
}
```

wsnyder, 2021-06-20:
I wouldn't have thought SystemC would tolerate that, but glad it does.  Can you make a pull request so I can look at the output directly please?

wsnyder, 2021-06-20:
P.S. Fine if the regressions don't pass.

gezalore, 2021-06-30:
Implemented by #3036

closed 2021-06-30 by gezalore