verilator/verilator #3037: i386 build fails beginning with 4.204

yurivict, 2021-06-22:
```
In file included from ../V3Active.cpp:33:
In file included from ../V3Active.h:24:
In file included from ../V3Ast.h:25:
In file included from ../V3Number.h:24:
./V3Hash.h:37:14: error: constructor cannot be redeclared
    explicit V3Hash(size_t val)
             ^
./V3Hash.h:33:14: note: previous definition is here
    explicit V3Hash(uint32_t val)
             ^
./V3Hash.h:56:12: error: class member cannot be redeclared
    V3Hash operator+(size_t value) const { return *this + V3Hash(value); }
           ^
./V3Hash.h:54:12: note: previous definition is here
    V3Hash operator+(uint32_t value) const { return *this + V3Hash(value); }
           ^
./V3Hash.h:63:13: error: class member cannot be redeclared
    V3Hash& operator+=(size_t value) { return *this += V3Hash(value); }
            ^
./V3Hash.h:61:13: note: previous definition is here
    V3Hash& operator+=(uint32_t value) { return *this += V3Hash(value); }
            ^
In file included from ../Verilator.cpp:18:
In file included from ../V3Ast.h:25:
In file included from ../V3Number.h:24:
./V3Hash.h:37:14: error: constructor cannot be redeclared
    explicit V3Hash(size_t val)
             ^
./V3Hash.h:33:14: note: previous definition is here
    explicit V3Hash(uint32_t val)
             ^
./V3Hash.h:56:12: error: class member cannot be redeclared
    V3Hash operator+(size_t value) const { return *this + V3Hash(value); }
           ^
./V3Hash.h:54:12: note: previous definition is here
    V3Hash operator+(uint32_t value) const { return *this + V3Hash(value); }
           ^
./V3Hash.h:63:13: error: class member cannot be redeclared
    V3Hash& operator+=(size_t value) { return *this += V3Hash(value); }
            ^
./V3Hash.h:61:13: note: previous definition is here
    V3Hash& operator+=(uint32_t value) { return *this += V3Hash(value); }
            ^
```

Log: http://beefy17.nyi.freebsd.org/data/main-i386-default/p670676cc73e1_s15fccd86b9/logs/verilator-4.204.log (IPv6)

gezalore, 2021-06-22:
Looks like duplicate of #3018, could you confirm master builds fine please?

yurivict, 2021-06-22:
I'll retest with the next release, thank you,

closed 2021-06-22 by yurivict