verilator/verilator #3040: '--x-assign unique' in combination with a parameter containing X

jeras, 2021-06-23:
Using Git master.

I use X valued bits in address decoders to combine the address and the mask into a single parameter.
I had to fix some unrelated issue where I used X as a don't care and Verilator missed the bug due to optimizing X into '0.
I was able to catch some of this bugs by using '--x-assign 1'.
So I tried to use '--x-assign unique' to see if my code had more X related bugs.

I got the next error:
```
%Error: Internal Error: ../tbn/r5p_tb.sv:136:9: ../V3Localize.cpp:162: AstVarRef not under function
                                              : ... In instance r5p_tb.ls_dec.__Vxrand1
  136 |   .AS  ({ {1'b1, {DAW{1'bx}}} ,    
      |         ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
X assignment should not be used on *right operands* of *Wildcard equality operators*, since X and Z are treated as wildcards.

I tried to find other cases where X/Z are used as wildcards:
11.4.6 Wildcard equality operators
11.4.13 Set membership operator (it uses the `==?` operator)
12.5.4 Set membership case statement

This is a placeholder for now, I will try to prepare a proper regression test, checking both if the code compiles, and if it works as expected under various X assignment settings.

wsnyder, 2021-06-24:
The X replacement code is supposed to not replace the right hand of wildcards, but there's probably a bug.    Thanks for looking into making a test.

wsnyder, 2021-11-23:
Was fixed with #3106.

closed 2021-11-23 by wsnyder