verilator/verilator #3043: "Internal Error" when set some signals public_flat_rw

yanx21, 2021-06-24:
Hi:
    I'm using verilator-4.204, and here is my testcase:
```verilog
module pipe_signal #(
    parameter   BUS_WIDTH   = 4
)
(
    input   wire                    i_clk,

    input   wire    [BUS_WIDTH-1:0] i_signal,
    output  wire    [BUS_WIDTH-1:0] o_signal
);

reg [BUS_WIDTH-1:0] signal_ff[2];

always @ (*) begin
    signal_ff[0] = i_signal;
end

always @ (posedge i_clk) begin
    signal_ff[1] <= signal_ff[0];
end

assign o_signal = signal_ff[1];

endmodule

module sub (
    input   wire    [1:0]   i_status /*verilator public_flat_rw*/,
    input   wire    [488:0] i_data  ,

    output  wire    [1:0]   o_status,
    output  wire    [488:0] o_data
);

assign o_status = i_status;
assign o_data   = i_data;

endmodule

module top
(
    input   wire            i_clk   ,

    input   wire    [1:0]   i_status,
    input   wire    [488:0] i_data  ,

    output  wire    [1:0]   o_status,
    output  wire    [488:0] o_data
);
wire    [1:0]   status ;
wire    [488:0] data   ;

pipe_signal #(
    .BUS_WIDTH  (491)
) U_PIPE (
    .i_clk      (i_clk             ),
    .i_signal   ({i_status, i_data}),
    .o_signal   ({status, data}    )
);

sub U_SUB (
    .i_status   (status  ),
    .i_data     (data    ),

    .o_status   (o_status),
    .o_data     (o_data  )
);

endmodule
```
And verilator gives the Error:
```txt
%Error: Internal Error: t.v:22:28: ../V3Subst.cpp:141: Reading a word that was never assigned, or bad word #
   22 | assign o_signal = signal_ff[1];
      |                            ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
But it runs smoothly if I remove the metacomment **public_flat_rw** to signal `i_status` of module `sub`

wsnyder, 2021-07-07:
When I run "verilator -cc above_file.v" on the master it works for me.  Did you try it on git master? Are there special flags or something needed to see the problem?  What happens if you run with --debug?

yanx21, 2021-09-06:
The problem appeared in published version 4.204, And I tried it on the git master, and the the problem has gone.

closed 2021-09-06 by yanx21